scottlamb/moonfire-nvr #187: Camera: Unable to parse SDP: No origin line.

bmsleight, 2022-01-10:
Describe the bug
Unable to parse SDP: No origin line.

Suspect error in rtsp streamed to moon-fire

https://github.com/aler9/rtsp-simple-server/issues/8
https://github.com/aler9/rtsp-simple-server/commit/2a6050d9b25d0083d115bf3b36f39289a18462e3
https://github.com/aler9/rtsp-simple-server/issues/742

Stream can be read from ffprobe and zoneminder
Can stream to ffmpeg on two difference machine from camara. 

Does  not work-
    ffmpeg -re -rtsp_transport udp_multicast -i rtsp://10.3.2.205:554/live/ch00_1  -c copy /tmp/output.mp4
Works:-
    ffmpeg -re -rtsp_transport udp  -i rtsp://10.3.2.205:554/live/ch00_1  -c copy /tmp/output.mp4

To Reproduce
Steps to reproduce the behavior:

Add camera to moonfire-nvr config then  moonfire-nvr run

Expected behavior
Live stream to be captured/s

Server (please complete the following information):

    v0.7.1-20-g77775a8
    manually compiled

Platform:

    Linux, Pi, rasbian

Additional context
 Commands below

```console
pi@moonfire-nvr:~ $ moonfire-nvr config
I20220110 21:19:13.038 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.34.1
I20220110 21:19:13.077 main moonfire_db::db] Loading video sample entries
I20220110 21:19:13.077 main moonfire_db::db] Loaded 0 video sample entries
I20220110 21:19:13.077 main moonfire_db::db] Loading sample file dirs
I20220110 21:19:13.078 main moonfire_db::db] Loaded 1 sample file dirs
I20220110 21:19:13.078 main moonfire_db::db] Loading cameras
I20220110 21:19:13.078 main moonfire_db::db] Loaded 1 cameras
I20220110 21:19:13.078 main moonfire_db::db] Loading streams
I20220110 21:19:13.079 main moonfire_db::db] Loaded 1 streams
I20220110 21:19:13.079 main moonfire_db::db] Loading recordings for camera Shed to Back Door stream Main
I20220110 21:19:13.079 main moonfire_db::db] Loaded 0 recordings for camera Shed to Back Door stream Main
I20220110 21:19:34.097 main moonfire_db::db] Flush 1 (why: drop): no recording changes
pi@moonfire-nvr:~ $ moonfire-nvr run
I20220110 21:19:48.225 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.34.1
I20220110 21:19:48.270 main moonfire_db::db] Loading video sample entries
I20220110 21:19:48.270 main moonfire_db::db] Loaded 0 video sample entries
I20220110 21:19:48.270 main moonfire_db::db] Loading sample file dirs
I20220110 21:19:48.271 main moonfire_db::db] Loaded 1 sample file dirs
I20220110 21:19:48.271 main moonfire_db::db] Loading cameras
I20220110 21:19:48.271 main moonfire_db::db] Loaded 1 cameras
I20220110 21:19:48.271 main moonfire_db::db] Loading streams
I20220110 21:19:48.272 main moonfire_db::db] Loaded 1 streams
I20220110 21:19:48.272 main moonfire_db::db] Loading recordings for camera Shed to Back Door stream Main
I20220110 21:19:48.272 main moonfire_db::db] Loaded 0 recordings for camera Shed to Back Door stream Main
I20220110 21:19:48.272 main moonfire_nvr::cmds::run] Database is loaded.
I20220110 21:19:48.300 main moonfire_nvr::cmds::run] Directories are opened.
I20220110 21:19:48.300 main moonfire_nvr::cmds::run] Resolved timezone: Europe/London
I20220110 21:19:48.300 main moonfire_nvr::cmds::run] Starting syncer for path /media/nvr/sample
I20220110 21:19:48.321 main moonfire_db::db] Flush 1 (why: synchronous deletion): no recording changes
I20220110 21:19:48.322 main moonfire_nvr::cmds::run] Starting streamer for Shed to Back Door-main
I20220110 21:19:48.322 main moonfire_nvr::cmds::run] Ready to serve HTTP requests
I20220110 21:19:48.322 s-Shed to Back Door-main moonfire_nvr::streamer] Shed to Back Door-main: Opening input: rtsp://10.3.2.205:554/live/ch00_1
W20220110 21:19:48.346 s-Shed to Back Door-main moonfire_nvr::streamer] Shed to Back Door-main: sleeping for PT1S after error: [10.3.2.208:39064(me)->10.3.2.205:554@2022-01-10T21:19:48, 0@2022-01-10T21:19:48] Ok response to DESCRIBE CSeq=1: Unable to parse SDP: No origin line

Length: 350 (0x15e) bytes
0000:   76 3d 30 0d  0a 73 3d 73  74 72 65 61  6d 65 64 20   v=0..s=streamed 
0010:   62 79 20 74  68 65 20 6d  61 63 72 6f  2d 76 69 64   by the macro-vid
0020:   65 6f 20 72  74 73 70 20  73 65 72 76  65 72 0d 0a   eo rtsp server..
0030:   74 3d 30 20  30 0d 0a 61  3d 63 6f 6e  74 72 6f 6c   t=0 0..a=control
0040:   3a 2a 0d 0a  61 3d 72 61  6e 67 65 3a  6e 70 74 3d   :*..a=range:npt=
0050:   30 2d 0d 0a  61 3d 78 2d  71 74 2d 74  65 78 74 2d   0-..a=x-qt-text-
0060:   6e 61 6d 3a  73 74 72 65  61 6d 65 64  20 62 79 20   nam:streamed by 
0070:   74 68 65 20  6d 61 63 72  6f 2d 76 69  64 65 6f 20   the macro-video 
0080:   72 74 73 70  20 73 65 72  76 65 72 0d  0a 63 3d 49   rtsp server..c=I
0090:   4e 20 49 50  34 20 30 2e  30 2e 30 2e  30 0d 0a 6d   N IP4 0.0.0.0..m
00a0:   3d 76 69 64  65 6f 20 30  20 52 54 50  2f 41 56 50   =video 0 RTP/AVP
00b0:   20 39 36 0d  0a 62 3d 41  53 3a 35 30  30 0d 0a 61    96..b=AS:500..a
00c0:   3d 72 74 70  6d 61 70 3a  39 36 20 48  32 36 34 2f   =rtpmap:96 H264/
00d0:   39 30 30 30  30 0d 0a 61  3d 66 6d 74  70 3a 39 36   90000..a=fmtp:96
00e0:   20 70 72 6f  66 69 6c 65  2d 6c 65 76  65 6c 2d 69    profile-level-i
00f0:   64 3d 54 65  41 6f 3b 70  61 63 6b 65  74 69 7a 61   d=TeAo;packetiza
0100:   74 69 6f 6e  2d 6d 6f 64  65 3d 31 3b  73 70 72 6f   tion-mode=1;spro
0110:   70 2d 70 61  72 61 6d 65  74 65 72 2d  73 65 74 73   p-parameter-sets
0120:   3d 4a 30 33  67 4b 49 31  6f 42 51 42  62 6f 51 41   =J03gKI1oBQBboQA
0130:   41 41 77 41  42 41 41 41  44 41 43 67  50 46 43 4b   AAwABAAADACgPFCK
0140:   67 2c 4b 4f  34 42 4e 4a  49 3d 0d 0a  61 3d 63 6f   g,KO4BNJI=..a=co
0150:   6e 74 72 6f  6c 3a 74 72  61 63 6b 31  0d 0a         ntrol:track1..

(set environment variable RUST_BACKTRACE=1 to see backtraces)
^CI20220110 21:19:49.200 main moonfire_nvr::cmds::run] Received SIGINT; shutting down gracefully. Send another SIGINT or SIGTERM to shut down immediately.
I20220110 21:19:49.200 main moonfire_nvr::cmds::run] Shutting down streamers and syncers.
I20220110 21:19:49.346 s-Shed to Back Door-main moonfire_nvr::streamer] Shed to Back Door-main: shutting down
I20220110 21:19:49.347 main moonfire_nvr::cmds::run] Waiting for HTTP requests to finish.
I20220110 21:19:49.348 main moonfire_nvr::cmds::run] Waiting for TEARDOWN requests to complete.
I20220110 21:19:49.348 main moonfire_nvr::cmds::run] Exiting.
I20220110 21:19:49.372 main moonfire_db::db] Flush 2 (why: drop): no recording changes

pi@moonfire-nvr:~ $ cd moonfire-nvr/
pi@moonfire-nvr:~/moonfire-nvr $ git describe --tags
v0.7.1-20-g77775a8

pi@moonfire-nvr:~/moonfire-nvr $ ffmpeg -hide_banner -re -rtsp_transport udp_multicast -i rtsp://10.3.2.205:554/live/ch00_1  -c copy /tmp/output.mp4
[rtsp @ 0x21b0d20] Nonmatching transport in server reply
rtsp://10.3.2.205:554/live/ch00_1: Invalid data found when processing input
pi@moonfire-nvr:~/moonfire-nvr $ ffmpeg -hide_banner -re -rtsp_transport udp  -i rtsp://10.3.2.205:554/live/ch00_1  -c copy /tmp/output.mp4
[udp @ 0x165ca00] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[udp @ 0x166cca0] attempted to set receive buffer to size 393216 but it only ended up set as 360448
Input #0, rtsp, from 'rtsp://10.3.2.205:554/live/ch00_1':
  Metadata:
    title           : streamed by the macro-video rtsp server
  Duration: N/A, start: 0.650000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(progressive), 1280x720, 20 fps, 20 tbr, 90k tbn, 40 tbc
File '/tmp/output.mp4' already exists. Overwrite? [y/N] n
Not overwriting - exiting

pi@moonfire-nvr:~/moonfire-nvr $ ffprobe -hide_banner rtsp://10.3.2.205:554/live/ch00_1 
[udp @ 0xa571c0] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[udp @ 0xa67510] attempted to set receive buffer to size 393216 but it only ended up set as 360448
Input #0, rtsp, from 'rtsp://10.3.2.205:554/live/ch00_1':
  Metadata:
    title           : streamed by the macro-video rtsp server
  Duration: N/A, start: 0.640000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(progressive), 1280x720, 20 fps, 50 tbr, 90k tbn, 40 tbc
pi@moonfire-nvr:~/moonfire-nvr $ uname -a
Linux moonfire-nvr 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
pi@moonfire-nvr:~/moonfire-nvr $ 
```

scottlamb, 2022-01-10:
For my notes: what brand/model/version of camera is this?

> Suspect error in rtsp streamed to moon-fire

I agree. [RFC 8866 section 9](https://datatracker.ietf.org/doc/html/rfc8866#section-9) says the origin field is mandatory:

```
   ; SDP Syntax
   session-description = version-field
                         origin-field
                         session-name-field
                         [information-field]
                         [uri-field]
                         *email-field
                         *phone-field
                         [connection-field]
                         *bandwidth-field
                         1*time-description
                         [key-field]
                         *attribute-field
                         *media-description
```

and it's not in the hex dump you quoted, so the camera is broken. It's in good company. Literally every camera I've tried is violating some RFC, and my practice is to make them work anyway.

The error is coming from the `sdp-types` crate. [Last time](https://github.com/sdroege/sdp-types/issues/10) that crate errored out, @sdroege accepted my PR to make it more permissive. Would you like to write a PR to make origin optional, or would you prefer I do it?

scottlamb, 2022-01-10:
As a workaround in the meantime, `moonfire-nvr run --rtsp-library=ffmpeg` should work. I'm planning to remove that code path, though, so we should get this working with the default `--rtsp-library=retina`.

bmsleight, 2022-01-12:
sorry work, interrupting life. 
Workaround is good. 
I will look at upstream soon. (not used coded rust before), but may take a while.

scottlamb, 2022-01-12:
Sounds good. I'm happy to help as needed/desired.

scottlamb, 2022-03-02:
Are you still planning to look at this? I can do it if not, as a precondition of removing ffmpeg.

commit 9d70d687 (2022-03-08): fix #187 via a dependency upgrade

I did a full `cargo upgrade` and fixed what it broke:

*   a couple things for the latest protobuf 3.0 alphas
    (note alphas don't promise API stability)
*   new minimum supported Rust version

This should have some other nice effects: parking_lot now uses inline
assembler, tokio has gotten faster, etc.
closed 2022-03-08 by scottlamb