scottlamb/moonfire-nvr #192: Config - "Test" against Raspberry Pi Zero 2 W rtsp server (cVLC) fails:  "Protocol not supported"

jlpoolen, 2022-01-29:
I have a Raspbery Pi Zero 2W running Bullseye (courtesy of the generous folks at Q-Engineering: [ready-to-burn 64-bit image for Zero](https://qengineering.eu/install-64-os-on-raspberry-pi-zero-2.html)) where I am running the suggested RTSP server described in the documentation at https://www.raspberrypi.com/documentation/accessories/camera.html#rtsp.  Looks like the libcamera project pipes its output to "cvlc" which I understand is a command-line invocation of VLC without a window/gui.  So it looks like VLC is acting as the rtsp server being fed by libcamera-vid.

The camera is the official Raspberry Pi HQ 12 Megapixel (see specs below) with the standard lens.

My server command is:

    pi@raspberrypi:~ $ cat stream.sh
    #!/bin/bash
    #
    # added: 192.168.1.12 where nothing existed
    #
    libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264
    pi@raspberrypi:~ $

I am able to connect to this server using VLC on Windows; I have had 6 sessions running with barely noticeable difference, if any, on the Pi Zero where the CPU usage hovers around 55-63%.  Here is URL I used for VLC on Windows:
[image]

Imbued with success, my next step is to have an instance of moonfire-nvr try to connect and then I am going to see what quality of connection is maintained over time.

(Note: I compiled gstreamer (1.19.2) and its rtsp, but have been having problems with gstreamer and PiZero's libcamera, so i backed off and tried the official Raspberry Pi document, [_supra_](https://www.raspberrypi.com/documentation/accessories/camera.html#rtsp), which has a section for creating an RTSP server.)

I configured a camera to the PiZero2 and when I tried to test, I received this error message: "Protocol not supported"
[image]

Now, Raspberry's libcamera-vid  RTSP server does not have a user/password configuration in the proposed command line.  When I connected with VLC, no user or password was required.  So in Moonfire-nvr, I left the username and password variable empty and tried.  When I encountered the "Protocol not supported", I then put in some bogus values, jlpoole/bogus, to see if a different error message was triggered.  Nothing change, same error message.

The libcamera-vid code is at https://github.com/raspberrypi/libcamera-apps/blob/main/apps/libcamera_vid.cpp.  But, I suspect the problem really lies within VLC's implementation of an rtsp server.

**To Reproduce**
Steps to reproduce the behavior:
1. On a RPiZero2 run: libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264
2. Launch moonfire-nvr config
3. Go to configuration of camera and enter information, click Test
4. See error "Protocol not supported"

**Expected behavior**
Standard confirmation that the connection worked.

**Screenshots**
see above

**Server (please complete the following information):**
 
   jlpoole@ares /usr/local/src/moonfire-nvr.git/trunk/server/target/release $ nvr --version
   moonfire-nvr 0.7.1
   jlpoole@ares /usr/local/src/moonfire-nvr.git/trunk/server/target/release $

    Using Moonfire:

    ares /usr/local/src/moonfire-nvr.git # svn info
    Path: .
    Working Copy Root Path: /usr/local/src/moonfire-nvr.git
    URL: https://github.com/scottlamb/moonfire-nvr.git
    Relative URL: ^/
    Repository Root: https://github.com/scottlamb/moonfire-nvr.git
    Repository UUID: 69f685ad-12d6-831e-7d07-ce57b7db7bff
    Revision: 1354
    Node Kind: directory
    Schedule: normal
    Last Changed Author: scott.lamb
    Last Changed Rev: 1354
    Last Changed Date: 2021-10-27 14:28:44 -0700 (Wed, 27 Oct 2021)

    ares /usr/local/src/moonfire-nvr.git #

(I will update and recompile to get latest high watermark, but I suspect this is a VLC rtsp implementation issue.)
moonfire-nvr config log file
[moonfire-nvr_config_log.txt](https://github.com/scottlamb/moonfire-nvr/files/7963023/moonfire-nvr_config_log.txt)

**Camera (please complete the following information):**
Raspberry Pi High Quality HQ Camera - 12MP
[Product ID: 4561 ](https://www.adafruit.com/product/4561)

**Desktop (please complete the following information):**
 - OS: **Bullseye**
      pi@raspberrypi:~ $ uname -a
     Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
     pi@raspberrypi:~ $
     pi@raspberrypi:~ $ cat /etc/os-release
     PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
     NAME="Debian GNU/Linux"
     VERSION_ID="11"
     VERSION="11 (bullseye)"
     VERSION_CODENAME=bullseye
     ID=debian
     HOME_URL="https://www.debian.org/"
     SUPPORT_URL="https://www.debian.org/support"
     BUG_REPORT_URL="https://bugs.debian.org/"
     pi@raspberrypi:~ $

jlpoolen, 2022-01-29:
Updated my local copy and rebuilt so I'm current with high watermark.  Tried same steps and got same result.

    ares /usr/local/src/moonfire-nvr.git/trunk # svn info
    Path: .
    Working Copy Root Path: /usr/local/src/moonfire-nvr.git
    URL: https://github.com/scottlamb/moonfire-nvr.git/trunk
    Relative URL: ^/trunk
    Repository Root: https://github.com/scottlamb/moonfire-nvr.git
    Repository UUID: 69f685ad-12d6-831e-7d07-ce57b7db7bff
    Revision: 1370
    Node Kind: directory
    Schedule: normal
    Last Changed Author: dependabot.bot.
    Last Changed Rev: 1368
    Last Changed Date: 2022-01-22 01:25:11 -0800 (Sat, 22 Jan 2022)

    ares /usr/local/src/moonfire-nvr.git/trunk # server/target/release/moonfire-nvr --version
    moonfire-nvr 0.7.1
    ares /usr/local/src/moonfire-nvr.git/trunk #

scottlamb, 2022-01-29:
That button is still using ffmpeg rather than retina. You could try running Retina's `cargo run --example client mp4` and/or an actual `moonfire-nvr run` and see if the result is any different.

jlpoolen, 2022-01-29:
Created a fresh database and populated with the PiZero2 camera.  
Here's a log of the run... Retina looks to have had problems, too.
[moonfire_rustbacktrace_log.zip](https://github.com/scottlamb/moonfire-nvr/files/7963106/moonfire_rustbacktrace_log.zip)

scottlamb, 2022-01-29:
From those logs:

```
E20220128 22:17:35.159 tokio-runtime-worker moonfire_nvr] panic at '/home/jlpoole/.cargo/registry/src/github.com-1ecc6299db9ec823/retina-0.3.4/src/codec/h264.rs:593:57': called `Option::unwrap()` on a `None` value
```

That looks like a bug that @unrelentingtech just fixed in Retina. I haven't updated Moonfire NVR to use the new version yet.

scottlamb, 2022-01-29:
6ca9f45 may help.

jlpoolen, 2022-01-29:
I executed "svn up" around 10:00 a.m. PST 1/29/22 and then "cargo build" from within  /usr/local/src/moonfire-nvr.git/trunk/server .  It looks like I picked up your change 6ca9f45 as evidence by line 51 of Cargo.toml.

ares /usr/local/src/moonfire-nvr.git/trunk/server # svn info
Path: .
Working Copy Root Path: /usr/local/src/moonfire-nvr.git
URL: https://github.com/scottlamb/moonfire-nvr.git/trunk/server
Relative URL: ^/trunk/server
Repository Root: https://github.com/scottlamb/moonfire-nvr.git
Repository UUID: 69f685ad-12d6-831e-7d07-ce57b7db7bff
Revision: 1372
Node Kind: directory
Schedule: normal
Last Changed Author: scott.lamb
Last Changed Rev: 1372
Last Changed Date: 2022-01-28 22:49:41 -0800 (Fri, 28 Jan 2022)

ares /usr/local/src/moonfire-nvr.git/trunk/server # cat Cargo.toml | head -n 51 |tail -n 1
retina = "0.3.7"
ares /usr/local/src/moonfire-nvr.git/trunk/server #

Still getting errors.  script of session attached

[PiZero2_cvlc_rtsp_moonfire_2022012921000_log.zip](https://github.com/scottlamb/moonfire-nvr/files/7964290/PiZero2_cvlc_rtsp_moonfire_2022012921000_log.zip)

Also, here is the cargo build log that has some warnings:

    ares /usr/local/src/moonfire-nvr.git/trunk/server # cargo build
      Downloaded retina v0.3.7
      Downloaded 1 crate (250.5 KB) in 0.61s
       Compiling retina v0.3.7
    warning: an associated function with this name may be added to the standard library in the future
       --> db/db.rs:951:27
        |
    951 |         s.on_live_segment.retain_mut(|cb| cb(l.clone()));
        |                           ^^^^^^^^^^
        |
        = note: `#[warn(unstable_name_collisions)]` on by default
        = warning: once this associated item is added to the standard library, the ambiguity may cause an error or change in behavior!
        = note: for more information, see issue #48919 <https://github.com/rust-lang/rust/issues/48919>
        = help: call with fully qualified syntax `retain_mut(...)` to keep using the current method

    warning: field is never read: `flags`
       --> db/auth.rs:195:5
        |
    195 |     flags: i32, // bitmask of SessionFlag enum values
        |     ^^^^^^^^^^
        |
        = note: `#[warn(dead_code)]` on by default

    warning: field is never read: `domain`
       --> db/auth.rs:196:5
        |
    196 |     domain: Option<Vec<u8>>,
        |     ^^^^^^^^^^^^^^^^^^^^^^^

    warning: field is never read: `description`
       --> db/auth.rs:197:5
        |
    197 |     description: Option<String>,
        |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^

    warning: field is never read: `creation_password_id`
       --> db/auth.rs:200:5
        |
    200 |     creation_password_id: Option<i32>,
        |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    warning: field is never read: `creation`
       --> db/auth.rs:201:5
        |
    201 |     creation: Request,
        |     ^^^^^^^^^^^^^^^^^

    warning: field is never read: `revocation_reason_detail`
       --> db/auth.rs:205:5
        |
    205 |     revocation_reason_detail: Option<String>,
        |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    warning: `moonfire-db` (lib) generated 7 warnings
       Compiling moonfire-nvr v0.7.1 (/usr/local/src/moonfire-nvr.git/trunk/server)
        Finished dev [unoptimized + debuginfo] target(s) in 17.80s
    ares /usr/local/src/moonfire-nvr.git/trunk/server #

scottlamb, 2022-01-29:
That log message says `retina-0.3.4`. It hasn't picked up the commit in question. I think github's Subversion mirrors are not reliably up-to-date.

jlpoolen, 2022-01-29:
I'll clone anew and update when I have it running.

scottlamb, 2022-01-29:
Actually, on second look: the log says `retina-0.3.4` but the compile output above says `retina-0.3.7`. You ran `cargo build` which would have produced an unoptimized build in `target/debug/moonfire-nvr`, and the log was from `target/release/moonfire-nvr`. Is it possible you ran an old release build rather than also doing `cargo build --release` first?

jlpoolen, 2022-01-29:
I just did "cargo build".  Have now launched "cargo build --release" from within server directory and its building everything anew.  Good catch.

jlpoolen, 2022-01-29:
After rebuild, reran.  I full log captured by "script", but for simplicity, here is the type of error message:

Let me know if you want me to post the zip archive of the session.

        ares /usr/local/src/moonfire-nvr.git/trunk/server # RUST_BACKTRACE=1 ./target/release/moonfire-nvr run
        I20220129 13:01:00.749 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.35.5
        I20220129 13:01:00.903 main moonfire_db::db] Loading video sample entries
        I20220129 13:01:00.903 main moonfire_db::db] Loaded 0 video sample entries
        I20220129 13:01:00.903 main moonfire_db::db] Loading sample file dirs
        I20220129 13:01:00.904 main moonfire_db::db] Loaded 1 sample file dirs
        I20220129 13:01:00.904 main moonfire_db::db] Loading cameras
        I20220129 13:01:00.904 main moonfire_db::db] Loaded 1 cameras
        I20220129 13:01:00.904 main moonfire_db::db] Loading streams
        I20220129 13:01:00.904 main moonfire_db::db] Loaded 1 streams
        I20220129 13:01:00.904 main moonfire_db::db] Loading recordings for camera pizero stream Main
        I20220129 13:01:00.904 main moonfire_db::db] Loaded 0 recordings for camera pizero stream Main
        I20220129 13:01:00.904 main moonfire_nvr::cmds::run] Database is loaded.
        I20220129 13:01:00.982 main moonfire_nvr::cmds::run] Directories are opened.
        I20220129 13:01:00.982 main moonfire_nvr::cmds::run] Resolved timezone: US/Pacific
        I20220129 13:01:00.982 main moonfire_nvr::cmds::run] Starting syncer for path /home/jlpoole/video/main
        I20220129 13:01:01.038 main moonfire_db::db] Flush 1 (why: synchronous deletion): no recording changes
        I20220129 13:01:01.038 main moonfire_nvr::cmds::run] Starting streamer for pizero-main
        I20220129 13:01:01.039 main moonfire_nvr::cmds::run] Ready to serve HTTP requests
        I20220129 13:01:01.039 s-pizero-main moonfire_nvr::streamer] pizero-main: Opening input: rtsp://192.168.1.12:8554/stream1
        W20220129 13:01:01.066 tokio-runtime-worker retina::codec::h264] Ignoring bad H.264 format-specific-params "packetization-mode=1;profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0IAAAAMAgAAAHnJQABMSwAAvrze9wDxImoA=,KO4fLA==;": key without value
        W20220129 13:01:01.122 s-pizero-main moonfire_nvr::streamer] pizero-main: sleeping for PT1S after error: [192.168.1.81:41894(me)->192.168.1.12:8554@2022-01-29T13:01:01, 678@2022-01-29T13:01:01] Unsupported Transport response to SETUP CSeq=2: Unexpected RTSP response status

        Backtrace:
           0: failure::backtrace::internal::InternalBacktrace::new
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/internal.rs:46:44
           1: failure::backtrace::Backtrace::new
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:121:35
              <failure::backtrace::Backtrace as core::default::Default>::default
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:140:13
           2: <failure::error::error_impl::ErrorImpl as core::convert::From<F>>::from
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/error_impl.rs:19:17
              <failure::error::Error as core::convert::From<F>>::from
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/mod.rs:36:18
              <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
                     at /rustc/1.58.0/library/core/src/result.rs:1911:27
              moonfire_nvr::stream::RetinaOpener::play::{{closure}}
                     at src/stream.rs:428:9
           3: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
                     at /rustc/1.58.0/library/core/src/future/mod.rs:80:19
              <tokio::time::timeout::Timeout<T> as core::future::future::Future>::poll
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.12.0/src/time/timeout.rs:149:33
              <moonfire_nvr::stream::RetinaOpener as moonfire_nvr::stream::Opener>::open::{{closure}}
                     at src/stream.rs:327:21
              <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
                     at /rustc/1.58.0/library/core/src/future/mod.rs:80:19
              tokio::runtime::task::core::CoreStage<T>::poll::{{closure}}
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.12.0/src/runtime/task/core.rs:161:17
              tokio::loom::std::unsafe_cell::UnsafeCell<T>::with_mut
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.12.0/src/loom/std/unsafe_cell.rs:14:9
           4: tokio::runtime::task::core::CoreStage<T>::poll
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.12.0/src/runtime/task/core.rs:151:13
              tokio::runtime::task::harness::poll_future::{{closure}}
                     at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.12.0/src/runtime/task/harness.rs:461:19
              <core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once
[... cut ...]
        I20220129 13:01:52.564 s-pizero-main moonfire_nvr::streamer] pizero-main: shutting down
        I20220129 13:01:52.564 main moonfire_nvr::cmds::run] Waiting for HTTP requests to finish.
        I20220129 13:01:52.564 main moonfire_nvr::cmds::run] Waiting for TEARDOWN requests to complete.
        I20220129 13:01:52.564 main moonfire_nvr::cmds::run] Exiting.
        I20220129 13:01:52.765 main moonfire_db::db] Flush 2 (why: drop): no recording changes
        ares /usr/local/src/moonfire-nvr.git/trunk/server # 

Let me know if you want me to post the script log.

scottlamb, 2022-01-29:
`W20220129 13:01:01.122 s-pizero-main moonfire_nvr::streamer] pizero-main: sleeping for PT1S after error: [192.168.1.81:41894(me)->192.168.1.12:8554@2022-01-29T13:01:01, 678@2022-01-29T13:01:01] Unsupported Transport response to SETUP CSeq=2: Unexpected RTSP response status`

I think `Unsupported Transport` means that Moonfire NVR requested TCP and the server only supports UDP (or vice versa). You could try flipping this option in the config.

`Protocol not supported` might be ffmpeg's way of saying the same thing.

jlpoolen, 2022-01-29:
I do not know what you are referencing in "the config.".  Moonfire's config??  If so, I ran in config mode and did not see anything in the cameras setup that specifies TCP vs. UDP.  Or do you mean the cvlc command line serving up the stream?  I tried adding "--sout-rtp-proto tcp " as follows:

     libcamera-vid -t 0 --inline -n --info-text "PiZero2 `date`" -o - | cvlc stream:///dev/stdin  --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' --sout-rtp-proto tcp :demux=h264

But no luck.

jlpoolen, 2022-01-29:
I also replaced "tcp" with "upd" and tried, still the same error.

    I20220129 13:49:02.967 s-pizero-main moonfire_nvr::streamer] pizero-main: Opening input: rtsp://192.168.1.12:8554/stream1
    W20220129 13:49:02.993 tokio-runtime-worker retina::codec::h264] Ignoring bad H.264 format-specific-params "packetization-mode=1;profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0IAAAAMAgAAAHnJQABMSwAAvrze9wDxImoA=,KO4fLA==;": key without value
    W20220129 13:49:03.018 s-pizero-main moonfire_nvr::streamer] pizero-main: sleeping for PT1S after error: [192.168.1.81:42052(me)->192.168.1.12:8554@2022-01-29T13:49:02, 678@2022-01-29T13:49:03] Unsupported Transport response to SETUP CSeq=2: Unexpected RTSP response status

    Backtrace:
       0: failure::backtrace::internal::InternalBacktrace::new
                 at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/internal.rs:46:44
       1: failure::backtrace::Backtrace::new
                 at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:121:35
          <failure::backtrace::Backtrace as core::default::Default>::default
                 at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:140:13
       2: <failure::error::error_impl::ErrorImpl as core::convert::From<F>>::from
                 at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/error_impl.rs:19:17
          <failure::error::Error as core::convert::From<F>>::from
                 at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/mod.rs:36:18
          <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
                 at /rustc/1.58.0/library/core/src/result.rs:1911:27
          moonfire_nvr::stream::RetinaOpener::play::{{closure}}
                 at src/stream.rs:428:9

jlpoolen, 2022-01-29:
Some lines from the extended help listing reference LIVE555

    pi@raspberrypi:~ $ vlc  --longhelp --advanced |cat -n |grep -i live555
    VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
      2262   RTP/RTSP/SDP demuxer (using Live555) (live555)
      3063        --demux {any,mp4,avi,asf,flacsys,es,mpc,tta,nuv,wav,caf,sid,sap,live555,mkv,diracsys,ogg,smf,adaptive,webvtt,ttml,aiff,rawvid,au,pva,ts,directory,xa,nsv,image,voc,mod,ps,ty,h26x,h26x,dvdnav,libbluray,mjpeg,mpgv,demux_cdg,rawdv,nsc,avformat,demux_stl,ps,vobsub,linsys_sdi,webvtt,subtitle,real,rawaud,es,demuxdump,vc1,stats,none}
    pi@raspberrypi:~

Moreover, included in the current code for VLC is https://code.videolan.org/videolan/vlc/-/blob/master/modules/access/live555.cpp

I'm guessing that VLC uses, perhaps exclusively, LIVE555 for its rtsp server.  That may not have any bearing on what you are considering, but I thought I'd bring it to your attention as you had taken a dive into LIVE555's code to form an opinion. Perhaps the inconsistent handshake between the cVLC piped content from Pi Zero and to Moonfire may be a caused by something you might remember.  A shot in the dark.

jlpoolen, 2022-01-29:
Also, this venture into using the libcamera pipe into cVLC was becuase I was having problems with gstreamer and when I saw this was a supported activity, I thought I'd try it to confirm it can be done.  Also, my foray into Gentoo is on hold because the initial boot-up files for Raspberry Pi products are proprietary -- i was hoping U-Boot could replace, and I'm seeing a Gentoo on Raspberry Pi kind of struggling, so I didn't want to take on a 10 hour investigation if I could finesse getting an rtsp server sending to Moonfire from a Raspberry Pi Zero 2 board on line asap to evaluate drop-outs and other anomalies I have experienced with Reolink cameras.  So now high priority on this issue, I'll probably go back to trying to get gStreamer on Bullseys working, even if with an old USB Logitech camera just so I can start sending packets to moonfire-nvr.  (I have discovered on the Pi Zero that having a display preview while streaming causes problems and that one is best advised to run an rstp server with the --no-preview ('-n') option.)

scottlamb, 2022-01-29:
> I do not know what you are referencing in "the config.". Moonfire's config??

Yes, the `rtsp_transport` option:

[image]

`(default)` means as specified on the commandline (`--rtsp-transport` command line option which in turns defaults to `tcp`). I added a config option recently so you can specify a different transport for different cameras.

jlpoolen, 2022-01-29:
Thanks... that line seemed invisible to me.  I tried all three states: 1) default, 2) tcp, and 3) udp.  I ran moonfire-nvr for 4 sets of error messages each and then terminated concluding the rtsp_transport variable did not make a difference.  (I reset the camera to default lest I forget in the future.)

scottlamb, 2022-01-31:
I tried this myself.

With TCP, I got the same behavior you described.

With UDP, I saw a different error: `Ok response to SETUP CSeq=2: Transport header is missing server_port parameter`. Moonfire/Retina is sending this:

```
Transport: RTP/AVP/UDP;client_port=22660-22661
```

then VLC is sending this:

```
Transport: RTP/AVP/UDP;destination=127.0.0.1;port=1234-1235;ttl=1;mode=play
```

The RFC says that `port` (rather than `server_port`) is for multicast. Apparently we were supposed to specifically request `unicast` if we wanted it:

```
   unicast | multicast:
          mutually exclusive indication of whether unicast or multicast
          delivery will be attempted. Default value is multicast.
          Clients that are capable of handling both unicast and
          multicast transmission MUST indicate such capability by
          including two full transport-specs with separate parameters
          for each.
```

...and we do want it. Retina doesn't support multicast. I accidentally left the `unicast` out of the transport spec for UDP. (It was there for TCP.) That might affect other servers, and it's easy to fix.

After that, there's some other error, but I might as well fix this one anyway...

jlpoolen, 2022-02-08:
A bug in libcamera was resolved ( [BUG] libcamera-vid piped into cVLC runs then crashes [#238](https://github.com/raspberrypi/libcamera-apps/issues/238) ), so I'm back to streaming using clvc piped in from libcamera-vid.

I started with a fresh pull of Retina and still error out.  I wasn't sure if the commit [1a00b5e](https://github.com/scottlamb/retina/commit/1a00b5e92a04a2bc2f9db2435f3c02cfdae92257) was to fix the problem.

    ares /usr/local/src # rm -r retina
    ares /usr/local/src # git clone https://github.com/scottlamb/retina.git
    Cloning into 'retina'...
    remote: Enumerating objects: 903, done.
    remote: Counting objects: 100% (903/903), done.
    remote: Compressing objects: 100% (561/561), done.
    remote: Total 903 (delta 601), reused 607 (delta 310), pack-reused 0
    Receiving objects: 100% (903/903), 461.43 KiB | 1.74 MiB/s, done.
    Resolving deltas: 100% (601/601), done.
    ares /usr/local/src # cd retina/
    ares /usr/local/src/retina # ls
    CHANGELOG.md  Cargo.lock  Cargo.toml  LICENSE-APACHE.txt  LICENSE-MIT.txt  README.md  benches  examples  fuzz  src
    ares /usr/local/src/retina #
 
8:03 PM 2/7/2022

    date; cargo run --example client mp4 --url rtsp://192.168.1.12:8554/stream1   /tmp/retina_out.mp4;date;
    
    
    ares /usr/local/src/retina # date; cargo run --example client mp4 --url rtsp://192.168.1.12:8554/stream1   /tmp/retina_out.mp4;date;
    Mon Feb  7 20:03:18 PST 2022
      Downloaded clap v2.34.0
      Downloaded tinyvec v1.5.1
      Downloaded ansi_term v0.12.1
      Downloaded rustc_version v0.4.0
      Downloaded semver v1.0.4
      Downloaded half v1.8.2
      Downloaded http-auth v0.1.4
      Downloaded plotters-svg v0.3.1
      Downloaded tokio-macros v1.6.0
      Downloaded serde_cbor v0.11.2
      Downloaded minimal-lexical v0.2.1
      Downloaded serde_json v1.0.72
      Downloaded futures v0.3.18
      Downloaded futures-macro v0.3.18
      Downloaded futures-executor v0.3.18
      Downloaded ryu v1.0.6
      Downloaded plotters-backend v0.3.2
      Downloaded futures-task v0.3.18
      Downloaded futures-sink v0.3.18
      Downloaded crossbeam-deque v0.8.1
      Downloaded criterion-plot v0.4.4
      Downloaded futures-channel v0.3.18
      Downloaded anyhow v1.0.51
      Downloaded futures-io v0.3.18
      Downloaded futures-core v0.3.18
      Downloaded cast v0.2.7
      Downloaded nom v7.1.0
      Downloaded tokio-util v0.6.9
      Downloaded futures-util v0.3.18
      Downloaded libc v0.2.108
      Downloaded tokio v1.14.0
      Downloaded syn v1.0.82
      Downloaded criterion v0.3.5
      Downloaded 33 crates (2.8 MB) in 1.09s
       Compiling cfg-if v1.0.0
       Compiling autocfg v1.0.1
       Compiling libc v0.2.108
       Compiling proc-macro2 v1.0.32
       Compiling unicode-xid v0.2.2
       Compiling version_check v0.9.3
       Compiling syn v1.0.82
       Compiling memchr v2.4.1
       Compiling lazy_static v1.4.0
       Compiling scopeguard v1.1.0
       Compiling serde v1.0.130
[... 114 lines cut ...]
       Compiling pin-project-internal v1.0.8
       Compiling structopt-derive v0.4.18
       Compiling structopt v0.3.25
       Compiling thiserror v1.0.30
       Compiling pin-project v1.0.8
       Compiling futures-executor v0.3.18
       Compiling futures v0.3.18
       Compiling tokio-util v0.6.9
       Compiling criterion v0.3.5
       Compiling retina v0.3.7 (/usr/local/src/retina)
        Finished dev [unoptimized + debuginfo] target(s) in 33.79s
         Running `target/debug/examples/client mp4 --url 'rtsp://192.168.1.12:8554/stream1' /tmp/retina_out.mp4`
    W20220207 20:03:52.524 main retina::codec::h264] Ignoring bad H.264 format-specific-params "packetization-mode=1;profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0IAAAAMAgAAAHnJQABMSwAAvrze9wDxImoA=,KO4fLA==;": key without value
    I20220207 20:03:52.525 main client::mp4] Using h264 video stream
    E20220207 20:03:52.549 main client] Fatal: [192.168.1.81:42196(me)->192.168.1.12:8554@2022-02-07T20:03:52, 678@2022-02-07T20:03:52] Unsupported Transport response to SETUP CSeq=2: Unexpected RTSP response status
    Mon Feb  7 20:03:52 PST 2022
    ares /usr/local/src/retina # date; RUST_BACKTRACE=1 cargo run --example client mp4 --url rtsp://192.168.1.12:8554/stream1   /tmp/retina_out.mp4;date;
    Mon Feb  7 20:05:32 PST 2022
        Finished dev [unoptimized + debuginfo] target(s) in 0.07s
         Running `target/debug/examples/client mp4 --url 'rtsp://192.168.1.12:8554/stream1' /tmp/retina_out.mp4`
    W20220207 20:05:32.819 main retina::codec::h264] Ignoring bad H.264 format-specific-params "packetization-mode=1;profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQFAe0IAAAAMAgAAAHnJQABMSwAAvrze9wDxImoA=,KO4fLA==;": key without value
    I20220207 20:05:32.819 main client::mp4] Using h264 video stream
    E20220207 20:05:32.844 main client] Fatal: [192.168.1.81:42198(me)->192.168.1.12:8554@2022-02-07T20:05:32, 678@2022-02-07T20:05:32] Unsupported Transport response to SETUP CSeq=2: Unexpected RTSP response status
    Mon Feb  7 20:05:32 PST 2022
    ares /usr/local/src/retina #