scottlamb/moonfire-nvr #204: Install.md doesn't produce working installation

ayounggun, 2022-03-11:
**Describe the bug**
I can't get moonfire-nvr working by following install.md.

I'm using a raspberry pi 4 8GB with a clean installation of archlinux armv8.

```
[tech@alarmpi alarm]$ nvr run
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Run a command in a new container

[tech@alarmpi alarm]$ sudo docker run --rm -it moonfire-nvr:latest
Unable to find image 'moonfire-nvr:latest' locally
docker: Error response from daemon: pull access denied for moonfire-nvr, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

[tech@alarmpi alarm]$ docker run --rm -it scottlamb/moonfire-nvr:latest
moonfire-nvr 0.7.1
security camera network video recorder

USAGE:
    moonfire-nvr <SUBCOMMAND>

FLAGS:
    -h, --help       Prints help information
    -V, --version    Prints version information

docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES

docker images
scottlamb/moonfire-nvr   latest    ff852895bc4b   4 months ago   601MB
hello-world              latest    1ec996c686eb   5 months ago   4.85kB

nvr logs
Error: No such container: moonfire-nvr

```

I have recorded my commands below.

# create a normal linux user with sudo privileges, install docker
ssh alarm@alarmpi
su root
passwd
passwd alarm
pacman-key --init
pacman-key --populate archlinuxarm
pacman -Syyu sudo git docker usbutils --noconfirm
groupadd sudo
visudo
useradd -m tech
usermod -aG tech docker
usermod -aG sudo docker
passwd tech
userdel alarm
reboot
systemctl start docker
systemctl enable docker
exit
exit

# check docker works without sudo
ssh tech@alarmpi
docker run hello-world
9b157615502d: Pull complete 
Digest: sha256:4c5f3db4f8a54eb1e017c385f683a2de6e06f75be442dc32698c9bbe6c861edd
Status: Downloaded newer image for hello-world:latest
Hello from Docker!
This message shows that your installation appears to be working correctly.

sudo useradd --user-group --create-home --home /var/lib/moonfire-nvr moonfire-nvr
sudo vi /usr/local/bin/nvr
sudo chmod a+rx /usr/local/bin/nvr

# the nvr script doesn't work for me when suggested to run it on line 134 of install.md. Does sudo vi /etc/moonfire-nvr.json need to be created first?
```
nvr init
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /etc/moonfire-nvr.json.
sudo nvr init
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /etc/moonfire-nvr.json.
```

# setup hard drive for sample dir
```
lsblk -f
sudo fdisk /dev/sda
sudo mkfs.ext4 -T largefile -m 1 /dev/sda1
lsusb -t
sudo vi /etc/fstab
UUID=34b78f2a-7b3b-44fb-b0fb-5e318d8d0d38   /media/nvr   ext4    nofail,noatime,lazytime,data=writeback,journal_async_commit  0       2

sudo mkdir /media/nvr
mkdir: cannot create directory '/media/nvr/': No such file or directory
sudo mkdir -p /media/nvr
sudo mount /media/nvr
sudo install -d -o moonfire-nvr -g moonfire-nvr -m 700 /media/nvr/sample
```
# is the following command needed as it tells me this path already exists?
```
sudo mkdir /media/nvr/sample
mkdir: cannot create directory '/media/nvr/sample': File exists

sudo chown -R moonfire-nvr:moonfire-nvr /media/nvr
sudo vi /usr/local/bin/nvr
        --mount=type=bind,source=/media/nvr/sample,destination=/media/nvr/sample
sudo vi /etc/moonfire-nvr.json
sudo nvr init

```
Thanks!

scottlamb, 2022-03-11:
> the nvr script doesn't work for me when suggested to run it on line 134 of install.md. Does sudo vi /etc/moonfire-nvr.json need to be created first?

I guess it does! Sorry, `/etc/moonfire-nvr.json` is new as of yesterday, and the instructions haven't been tested much since.

> is the following command needed as it tells me this path already exists?

No, it's not. I'll remove it.

closed 2022-03-11 by scottlamb

scottlamb, 2022-03-11:
Oh! I almost forgot to point out something important:

The latest version Docker on the Docker hub is 0.7.1, which was before I added `/etc/moonfire-nvr.json`. So you should follow the [matching version](https://github.com/scottlamb/moonfire-nvr/blob/v0.7.1/guide/install.md) of the Docker instructions. The ones you're looking at now are instructions in progress for a future `v0.7.2` release...

Perhaps I should update `install.md` to look for this pull-down when viewing the instructions on github:

[image]

commit 0a9ec04a (2022-03-11): install instructions updates

Fixes #204

* The docker container won't start if a bind refers to a path that
  doesn't exist yet, so move the /etc/moonfire-nvr.json creation up
* Remove redundant command in the dedicated hard drive setup, and
  improve its clarity
commit de28f6ee (2022-03-11): point to correct version of install instructions

This is particularly important right now. The master branch describes
a config file that doesn't exist in the latest release (0.7.1).

Fixes #204