scottlamb/moonfire-nvr #205: Install.md tag 0.7.1 doesn't produce working installation

eggnog-armed, 2022-03-12:
Hi scottlamb

Thanks for your quick response before.
Perhaps I should try and earlier tag of install.md than 0.7.1 as its not working for me unfortunately.
I really like the look of this NVR so am keen to get it working!

**Description of Bug**
I can't get moonfire-nvr working by following install.md tag 0.7.1.
nvr run fails with the following
```
nvr run
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]
Run a command in a new container
```
I'm using a raspberry pi 2 with a clean installation of archlinux armv7.
I'm not familiar with docker so my apologies if I have missed something.
I think docker is not finding the moonfire-nvr image locally even through docker image ls shows it is there.
```
docker image ls
REPOSITORY               TAG       IMAGE ID       CREATED        SIZE
scottlamb/moonfire-nvr   latest    ff852895bc4b   4 months ago   601MB
hello-world              latest    1ec996c686eb   5 months ago   4.85kB
```

If I run the following command I get something more hopeful
```
docker run --rm -it scottlamb/moonfire-nvr:latest 
moonfire-nvr 0.7.1
security camera network video recorder

USAGE:
    moonfire-nvr <SUBCOMMAND>

FLAGS:
    -h, --help       Prints help information
    -V, --version    Prints version information

SUBCOMMANDS:
    check      Checks database integrity (like fsck)
    config     Interactively edits configuration
    help       Prints this message or the help of the given subcommand(s)
    init       Initializes a database
...
```

**To Reproduce**
run the following commands mainly from install.md tag 0.7.1:

```
sudo systemctl start docker
sudo systemctl enable docker

docker run hello-world

9b157615502d: Pull complete 
Digest: sha256:4c5f3db4f8a54eb1e017c385f683a2de6e06f75be442dc32698c9bbe6c861edd
Status: Downloaded newer image for hello-world:latest

Hello from Docker!
This message shows that your installation appears to be working correctly.

sudo useradd --user-group --create-home --home /var/lib/moonfire-nvr moonfire-nvr
sudo vi /usr/local/bin/nvr
sudo chmod a+rx /usr/local/bin/nvr

nvr init
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /etc/moonfire-nvr.json.
sudo nvr init
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /etc/moonfire-nvr.json.

sudo vi /etc/fstab
UUID=4babe2a4-0354-4d04-9b9c-e9fe611bb6bc /media/nvr ext4 nofail,noatime,lazytime,data=writeback,journal_async_commit 0 sudo mkdir -p /media/nvr
sudo mount /media/nvr
sudo install -d -o moonfire-nvr -g moonfire-nvr -m 700 /media/nvr/sample
sudo mkdir /media/nvr/sample
sudo chown -R moonfire-nvr:moonfire-nvr /media/nvr

nvr config 2>debug-log
cat debug-log 
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /etc/moonfire-nvr.json.
See 'docker run --help'.
```

0.7.1 still wants json because json is referenced in the 0.7.1 bash script I think.
I therefore switch to master install.md and create json.

```
sudo vi /etc/moonfire-nvr.json:

{
    "binds": [
        {
            "ipv4": "0.0.0.0:8080",
            "allowUnauthenticatedPermissions": {
                "viewVideo": true
            }
        },
        {
            "unix": "/var/lib/moonfire-nvr/sock",
            "ownUidIsPrivileged": true
        }
    ]
}

```
This allows nvr init to run successfully 
```
sudo nvr init
Unable to find image 'scottlamb/moonfire-nvr:latest' locally
latest: Pulling from scottlamb/moonfire-nvr
...
Status: Downloaded newer image for scottlamb/moonfire-nvr:latest
I20220312 09:50:11.098 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in Create mode with SQLite version 3.31.1
I20220312 09:50:11.275 main moonfire_nvr::cmds::init] Database initialized.

```
I then configure a sample folder, camera and user with the config tool successfully.
 nvr run
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Run a command in a new container
`sudo nvr config 2>debug-log`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server (please complete the following information):**
  - If using Docker: `docker ps` + `docker images`
  - If building from git: `git describe --dirty` + `moonfire-nvr --version`
  - Attach a [log file](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#viewing-moonfire-nvrs-logs). Run with the `RUST_BACKTRACE=1` environment variable set if possible.

**Camera (please complete the following information):**
  - Camera manufacturer and model: [e.g. Reolink RLC-410]
  - Firmware version: [e.g. `V2.0.0.1215_16091800`]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
I'm using a raspberry pi 2 with a clean installation of archlinux armv7.

scottlamb, 2022-03-14:
Thanks for reporting!

> `"docker run" requires at least 1 argument.`

Argh. The `/usr/local/bin/nvr` script on the install guide is broken. It has comments and blank lines in the middle of the `docker run` command, but they end the command prematurely. If you remove these lines, it should work.

Embarrassingly, it's been broken for months (since 30cea5c). Not only did I not notice (ideally I'd do a fresh install on each release to test the docs, but I haven't done it recently) but you're the first to mention it. Best guess is everyone else either already had their `nvr` script in place, are building from source (I know several people are using the build.md instructions instead), or noticed and fixed this on their own installs without saying anything.

Anyway, I'll fix the guide. As `v0.7.1`'s matching docs are broken, I think I'll roll a `v0.7.2` shortly and recommend that (rather than say having a branch for fixes to each release's docs). There are several fixes/improvements waiting for a release anyway.

The docker thing was an experiment to improve ease of installation, and it hasn't been totally successful. I'd like to switch to a zero-dependency precompiled binary (#160) but am not quite there yet.

umbrellarunaround, 2022-03-14:
Great thanks! that makes sense. I'm glad to help and will give it a go later tonight.

eggnog-armed, 2022-03-15:
Thanks I can confirm I can install moonfire nvr by removing the commented and blank lines.

closed 2022-03-15 by eggnog-armed