scottlamb/moonfire-nvr #225: Unable to resolve /etc/localtime symlink while starting server

katyo, 2022-05-09:
When I trying to start server on NixOS it fails with the following error:
```
E20220510 00:34:16.546 main moonfire_nvr] Exiting due to error: Unable to resolve /etc/localtime symlink destination /etc/zoneinfo/Asia/Yekaterinburg to a timezone.
```

On a NixOS localtime symlink resolves to something like:
```
$ realpath /etc/localtime
/nix/store/<SOME_HASH>-tzdata-<VERSION>/share/zoneinfo/<REGION>/<LOCATION>
```

But as I see current solution simply strips some well-known prefixes:
https://github.com/scottlamb/moonfire-nvr/blob/841e06e35458db1e40e16c72730cd2959cfdc5e6/server/src/cmds/run/mod.rs#L47-L48

scottlamb, 2022-05-09:
Thanks for the report. Maybe we should be looking mid-string for `/zoneinfo/` then rather than by prefix.

You can work around by setting the TZ environment variable to `:Asia/Yekaterinburg`. On most shells, `export TZ=:Asia/Yekaterinburg` will do.

katyo, 2022-05-10:
@scottlamb I created PR which implements that idea.

closed 2022-05-11 by katyo