scottlamb/moonfire-nvr #230: moonfire-nvr:v0.7.5 docker image is actually moonfire 0.7.4

IronOxidizer, 2022-06-23:
**Describe the bug**
moonfire-nvr:v0.7.5 docker image is actually moonfire 0.7.4

**To Reproduce**
Steps to reproduce the behavior:
1. Install moonfire-nvr:v0.7.5
2. ```
    >>> docker images
    scottlamb/moonfire-nvr   v0.7.5    bfff6303feba   6 weeks ago   308MB
    ```
3. ```
    >>> nvr --version
    moonfire-nvr 0.7.4
    ```

**Additional context**

Additionally, I'm pretty sure this isn't just a version misprint, but the actual incorrect build because https://github.com/scottlamb/moonfire-nvr/issues/219 is still an issue on this version, even though its supposed to be fixed in v0.7.5

scottlamb, 2022-06-24:
Oh no! thanks for the report; I'll try to figure out tonight what happened.

guptayash, 2022-07-10:
Hey, any updates on this? I am also facing the same issue. 
Thanks!

IronOxidizer, 2023-07-09:
Fixed in 0.7.6

closed 2023-07-09 by IronOxidizer