scottlamb/moonfire-nvr #244: Daylight Savings Snafu?

jlpoolen, 2022-11-06:
Preface: I have a Reolink RLC-420 which had an incorrect setting for the "set-back" time for Daylight savings time, the time was set back Oct 31st when it should have been set back on November 6th.  I was unaware of the incorrect setting until the filing of this issue. This is disclosed to avoid confusion when looking at screenshots herein.

[image]

On Sunday morning, November 6th (after Daylight Savings set back earlier in the day), I was reviewing Saturday, November 5th, recordings.  I had found a recording of interest which Reolink's software captured at 2:05 AM:

[image]

I use Moonfire-nvr to retrieve a full snapshot since Reolink's start and end time of snippets is invariably insufficient.  So I went to the Moonfire-nvr web interface to get a better snapshot of the event occurring at Saturday, 11/5 at  2:04 AM PDT.  I had performed a refresh of the web page as I usually do when I start a session to review recordings.

[image]

What is strange is that I specified a start and stop time of approximately 1 hour (1:05 - 2:06), yet 3 hours worth of video appear.  I took screenshots of each of the three videos shortly after they commenced running and aligned the screenshots with the table entries.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Moonfire-nvr portal
2. Set start and end times, 1 hour apart
2. See 3 hours worth of videos when there should only be one.

**Expected behavior**
A table of videos which total approximately 1 hour.

**Server (please complete the following information):**
  - If using Docker: **not** using Docker
  - If building from git: 
```
 jlpoole@pi:/usr/local/src/moonfire-nvr $ git describe --dirty
v0.7.4-3-gfd7438d-dirty
jlpoole@pi:/usr/local/src/moonfire-nvr $
```
```
jlpoole@pi:/usr/local/src/moonfire-nvr $ date; moonfire-nvr --version
Sun 06 Nov 2022 09:13:24 AM PST
moonfire-nvr 0.7.4
jlpoole@pi:/usr/local/src/moonfire-nvr $

```
  - Attach a [log file] -- I can stop and provide a log file if requests.

**Camera (please complete the following information):**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Browser: Firefox 
 - Version: 106.0.2 (64-bit) (needs to be updated)

**Additional context**
```
Running Moonfire-nvr on Raspberry Pi 4:
jlpoole@pi:~ $ date; uname -a
Sun 06 Nov 2022 08:53:56 AM PST
Linux pi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux
jlpoole@pi:~ $
```

```
jlpoole@pi:/usr/local/src/moonfire-nvr $ git log -1
commit fd7438dd2818469820547165a505519ebeadba58 (HEAD -> master, origin/master, origin/HEAD, show-current)
Author: Scott Lamb <slamb@slamb.org>
Date:   Wed Apr 13 21:45:45 2022 -0700

    ignore port number in ws origin check

    Fixes #219
```
The date on the Raspberry Pi is correct for my timezone:

```
jlpoole@pi:/usr/local/src/moonfire-nvr $

 date
Sun 06 Nov 2022 09:07:59 AM PST
jlpoole@pi:/usr/local/src/moonfire-nvr $

```

jlpoolen, 2022-11-06:
Here's a further twist, when I tried to specify a ~2 minute segment starting at 01:05, what is displayed is 00:05

[image]

Perhaps this is a browser cache issue or something happening on the JavaScript side?  I'll restart Firefox and attempt again.

scottlamb, 2022-11-06:
I'm on my phone right now and can't look closely, but it wouldn't surprise me to learn we have a time zone bug in the UI layer. The JavaScript time libraries available are really sketchy wrt time zone changes unfortunately.

jlpoolen, 2022-11-06:
Interesting, when I tried to focus on a 2 minute segment which should be at 1:05 AM PDT, Saturday 11/5, I entered a start time of 01:05:50 and an end time of 01:06:36.  What is displayed on the table are my specified times.

[image]

So, I appear to be off 1 hour based on the timestampe burned into the video.  I therefore try to compensate, and change my "01:" to "02:" on both start and end times.  What is then displayed in the table are values of 03 instead of 02.

[image]

Then I tried just changing the start time to "01:" so I have specified 1 hour + 1 minute, and then three table entries appear offering just over 2 hours instead of the expected 1 hour.  If I click the 2nd table entry, I can then access the video I want with the burned-in time of "01:05".

[image]

So, I'm almost at a "it works for me" work-around, the problem I still face is I like to be able to view a video, set my start and end times based upon what I actually want to preserve.  In this case, I can display the video, but it is in a huge segment, e.g. ~ 1 hour.  I want to be able to save only a 2 minute section.  I tried the Max Video settings, but found there to be a 1 hour minimum.

[image]

What would be helpful for me is the ability to extract a 0 minute 46 second video starting at the burned-in time of 1:05:50.  (Documents a guy hopping a fence to avoid detection of the security lights.)

scottlamb, 2022-11-06:
I'm at a computer now and see the same broken behavior on fall back day.

Fortunately the backend's time handling is solid. If I needed that clip right now, I would go to the video clip with the correct start time and save the URL via the browser's context menu. E.g., clicking here in Chrome:

[image]

and ending up with e.g. `https://nvr.home.slamb.org/api/cameras/225c6eeb-9f77-4ddf-ab8e-b5788642fa3c/main/view.mp4?s=2706893-2706958.3968834-354968834&ts=false` in the clipboard. Then I'd compare to the [API documentation for `view.mp4`](https://github.com/scottlamb/moonfire-nvr/blob/master/design/api.md#get-apicamerasuuidstreamviewmp4). Look at the part after the `.` here. It says it starts at 3,968,834 90,000ths after the start of segment 2,706,893 and continues until 354,968,834 90,000ths of a second after that same segment. If I wanted it to only be 46 seconds long, I'd take that first number, add 46 * 90,000, change the second number to that, and put that in the URL bar.

jlpoolen, 2022-11-06:
Thank you, Scott.  Yes, it looks like the video is there, just having a problem getting it accessible through the web browser.

I'm having problems following your instructions.  I tried with Firefox, but could not get a precision timecode, e.g. the 90k number ("90k Precision") following a decimal point, so I tried with Chrome and cannot discern what you mean by the "browser's context menu".  When I try copying the the Video Address, I do not get the 90k Precision.  Here's what goes into my clipboard when I Copy Video Address from Chrome:

http://pi:8080/api/cameras/f7689025-966d-4ca7-b045-5470ca33f6bf/main/view.mp4?s=421652-421708&ts=false

Now, I do see a listing via this URL:
http://pi:8080/api/cameras/f7689025-966d-4ca7-b045-5470ca33f6bf/main/recordings

So, I thought perhaps I might find the video file using the criteria of 421652, but no luck:

[image]

Could the fact that I am using a version of Moonfire-nvr as of April rather than the current high watermark be what is causing my URLs not to have high precision 90k seconds?  I've been _reticent to upgrade_ until I know I can commit a couple of hours or 1/2 day of potential downtime.  (We have a lot happening in Penny Lane camera coverage (5 cameras), including a Lady Godiva bicyclist who mooned the cameras in protest.  The story that is evolving about The Midnight Gardener, the person my cameras have been documenting as as possible suspect for theft, is becoming legend within our historic district and beyond.)

scottlamb, 2022-11-06:
The URL `http://pi:8080/api/cameras/f7689025-966d-4ca7-b045-5470ca33f6bf/main/view.mp4?s=421652-421708&ts=false` goes from the exact beginning of segment 421652 to the exact end of segment 421708. If you want it to go for e.g. 1 second, you can append a `.-90000` to the `s` parameter (`http://pi:8080/api/cameras/f7689025-966d-4ca7-b045-5470ca33f6bf/main/view.mp4?s=421652-421708.-90000&ts=false`) and it should do so.

> Could the fact that I am using a version of Moonfire-nvr as of April rather than the current high watermark be what is causing my URLs not to have high precision 90k seconds?

The URL format hasn't changed in a while. It's just about the clip you're starting from. I selected a specific start time that was in the middle of a recording, so mine had a `.blah-blah` already.

scottlamb, 2022-11-06:
Oh, here's an alternate workaround that may be easier: restart Moonfire with the environment variable `TZ` set to a time zone that doesn't do daylight saving time, e.g. `TZ=UTC`. No messing with URLs then.

jlpoolen, 2022-11-06:
Just to pin this down, I successfully accessed the 56 second segment I wanted by:

1) calculating 90,000 times the number of seconds I wanted.  I chose 56 for a product of: 5040000
2) inserted after the second number of the URL, i.e. 421708, a "." plus a minus "-" followed by the product: 5040000. 

So my url becomes:
http://pi:8080/api/cameras/f7689025-966d-4ca7-b045-5470ca33f6bf/main/view.mp4?s=421652-421708.-5040000&ts=false

You're modify the URL by inserting a dot followed by a negative number.

Thank you, Scott.

jlpoolen, 2022-11-06:
> Oh, here's an alternate workaround that may be easier: restart Moonfire with the environment variable `TZ` set to a time zone that doesn't do daylight saving time, e.g. `TZ=UTC`. [Edit: and refresh the browser window.] No messing with URLs then.

The downside is that if you are sharing the site with people not familiar with timezone differences, this could confuse them.  I share my site with people who are not technically inclined, so keeping the time to Pacific Time removes an obstacle.

scottlamb, 2022-11-07:
Glad you got what you needed for now.

I'd love to fix this properly, even to the level of having a date time chooser that lets you disambiguate the fall back hour (2am to 3am) in a slick way. It's not easy right now though with the typically used libraries. The JS Temporal API being standardized now will provide a firmer foundation at least. Even then though I don't know if the mui folks whose datetime picker component I'm using care about this level of detail. Before the latest UI require, I did have a way you could type a zone offset into the time text box. Maybe it wouldn't be too hard to reintroduce that.

robin-thoni, 2025-10-26:
Looks like I'm experiencing a similar issue:

[image]

I'm selecting `19:50:00` as end time, but I'm really getting `20:50:00`. Interestingly, I'm not even selecting a range over the time switch, but fully before, while we're currently right after the DST switch.

(BTW, the displayed times over the DST switch are a bit confusing too:

[image])