scottlamb/moonfire-nvr #252: User Access Control Structure?

scottlamb, 2022-12-24:
### Discussed in https://github.com/scottlamb/moonfire-nvr/discussions/236

<div type='discussions-op-text'>

<sup>Originally posted by **jlpoolen** August  9, 2022</sup>
I know this is way beyond the design criteria, but I am pleased to say that I have been opening my instance of moonfire-nvr to neighbors (using pfSense and NAT routing port 8080 to my raspberry pi) and nearby property owners so they may access the feeds, live and historic, directly.  I create an account for each user.  So the use of the server has become more than a single user/owner.  One neighbor who monitors further west the portion of the alley my cameras cover constantly logs in when something suspicious comes by his cameras (Swann brand) and he wants to get a better shot.  I run my cameras at maximum resolution and frames per second to get the best possible image.  I'm constantly being asked about my system because the quality of the Reolinks at full resolution gives a remarkable picture.  

It would be great if there could be some sort of access control list for each user that might specify what cameras/feeds they may access.  I currently have 11 cameras and will be adding two more, one to cover an adjacent property owner's lot bringing the count to two of cameras devoted to other premises.

I appreciate how complicated access control lists, aka ACLs, can be and can sometimes be counter-productive.

The camera set-up I have coupled with Moonfire-nvr has helped tremendously to discourage problem people - transients using the alley as a bathroom, illegal dumping (small scale, but dumping nonetheless) and drug activity.  We had one problem woman who appears to be stealing gardening stuff, plants, a rose arbor, flowers, and she would ride by between midnight and 6:00 a.m. with her booty, I published a series of her passing by and called her The Midnight Gardener and shared the select videos with neighbors.  Finally a neighbor contacted her landlord as her activity was escalating and the landlord confronted her after he demanded proof and then viewed the videos.  Needles to say, she rarely uses the alley and coincidentally there appears to be a significant reduction of what looks like drug transactions.  I think word got out on the street that the cameras do, indeed, capture everything.  Everyone agrees the alley has improved thanks to my and a couple of neighbors being vigilant with our surveillance.  It is the ability to share the Moonfire-nvr interface via a browser that makes this project sparkle.

I'm starting to sound like a constant cheerleader, but when there is something to cheer about, the cheerleaders cheer.</div>

scottlamb, 2022-12-24:
I'm going to implement basic support for this. I'll allow embedding a list of cameras into the permissions, and (on the fresh and unmerged `user-admin` branch) you can adjust permissions via API endpoints. I don't think I'll support this finer-grained permission stuff in the `moonfire-nvr config` TUI interface, which I want to get rid of. I'll eventually support it in the pretty HTTP UI but not right away, so in the meantime you'll have to send HTTP requests programmatically or on the commandline with `curl` or whatever to configure this.

How would you expect the new permission to work?

Right now the global permission `view_video` really only controls viewing video—without this permission, you can still see the cameras and know when they have recorded video and whatnot, but a request to view video will fail with a 403 error (and the UI doesn't handle this well). I could keep that and add `view_video_for_camera_ids` that allows viewing video for just those camera ids, but I think this behavior may not be what anyone wants.

So...if you are allowing someone to only view video for camera A, should they be able to see camera B in the UI at all? Should they know when there's recorded video on camera B, when there's recorded motion, etc.? I could add a fine-grained permission to control each of these things but it might be better to start simple.

scottlamb, 2022-12-24:
@jlpoolen

jlpoolen, 2023-02-11:
What about having something associated with a users account such as an array of cameras they are associated with.  That array could then be referenced when generating the UI so that only cameras, or their ids, associated with their account are displayed.  

Another approach might  be to have more than one server instance, port 8082,8083 and the owner could determine which cameras are displayed on which instance?

scottlamb, 2023-02-16:
So users don't see cameras they don't have access to in the list at all? Yeah, that sounds pretty reasonable.

> Another approach might be to have more than one server instance, port 8082,8083 and the owner could determine which cameras are displayed on which instance?

Also possible but seems like a worse experience. It's nice to just expose a port to the Internet and have Moonfire's own user/session auth be sufficient.