scottlamb/moonfire-nvr #256: User Interface: Time Codes Constrained?

jlpoolen, 2023-01-14:
Under prior versions, I could select a start and stop time down to the second, that allowed me to determine the precise stop/start points of videos which I could then download.

Now when I try to change the numbers, i.e. 09:28:25, in the first "Time" field, the interface will not allow me to specify the seconds, e.g. 25, but reverts to another number, e.g. 01.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Moonfire main web page
2. Select camera, and then try to enter a date in the first "Time" field
3. "Time" field will not accept for input, or allow me to change the seconds field to what I want, e.g. 25.

**Expected behavior**
The ability to set HH:MM:SS to the second.

**Screenshots**
[image]

**Server (please complete the following information):**
```
jlpoole@pi:/usr/local/src/moonfire-nvr $ date; git describe --dirty; moonfire-nvr --version
Sat 14 Jan 2023 12:02:33 PM PST
v0.7.5-38-g2667dd6
moonfire-nvr 0.7.5
jlpoole@pi:/usr/local/src/moonfire-nvr $
```

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Browser Firefox
 - Version 108.0.2

**Additional context**
I upgraded recently and just noticed this change in behavior.

The ability to select start and stop times is extremely helpful in creating videos which do not have any wasted time before or after the desired action being documented.

jlpoolen, 2023-01-14:
Also, I've tried using destructive backspace and the arrow key to alter the seconds ("SS") value and the field will not let me do so.
[image]

scottlamb, 2023-02-13:
Thanks. This looks like a problem in the `@mui/x-date-pickers` component I'm using; I reported it [there](https://github.com/mui/mui-x/issues/7935). I've noticed that while you can't backspace, you can use the left arrow and then type over the existing seconds value.

jlpoolen, 2023-02-13:
The work-around is to create the date sequence, i.e.  10:20:03, in a text editor and then copy paste the value into the field.

jlpoolen, 2023-10-14:
Just a heads up: I upgraded today, October 14, 2023, and the problem with the seconds field not accepting what I key in when just the two seconds digits are highlighted continues.

jlpoolen, 2023-10-14:
> Thanks. This looks like a problem in the `@mui/x-date-pickers` component I'm using; I reported it [there](https://github.com/mui/mui-x/issues/7935). I've noticed that while you can't backspace, you can use the left arrow and then type over the existing seconds value.

Somehow I missed your suggested left-arrow work-around and tried it successfully now; it worked.  Thank you!

commit 1f780610 (2023-11-28): upgrade `@mui/x-date-pickers` to v6 beta

Fixes #256
closed 2023-11-28 by scottlamb