scottlamb/moonfire-nvr #271: Won't boot: "unknown field `allowUnauthenticatedPermissions` for key `binds` "

kerryland, 2023-06-12:
I'm kicking the tires and have just done a fresh install, but she won't start up.

Logs tell me:
```
E20230612 16:20:57.372 main moonfire_nvr] Exiting due to error: unable to read /etc/moonfire-nvr.toml
caused by: unknown field `allowUnauthenticatedPermissions` for key `binds` at line 5 column 1
```

My `/etc/moonfire-nvr.toml` file looks like this:
```
[[binds]]
ipv4 = "0.0.0.0:8080"
allowUnauthenticatedPermissions = { viewVideo = true }

[[binds]]
unix = "/var/lib/moonfire-nvr/sock"
ownUidIsPrivileged = true
```

I'm not sure what the next step is.

scottlamb, 2023-07-05:
Apologies for missing this issue until now.

I think the problem is that you're using the latest released version but the instructions from unreleased HEAD. There have been changes in the config format between the two.

You can either look at the `v0.7.5` instructions by picking that tag in the pull-down...

[image]

...or wait a little bit; I'm going to release a `v0.7.6` within the next day or two.

kerryland, 2023-07-05:
I can wait -- thanks!

closed 2023-07-05 by kerryland

scottlamb, 2023-07-08:
Bit delayed, but 0.7.6 is out now.