scottlamb/moonfire-nvr #285: Exiting due to error: unable to read /etc/moonfire-nvr.toml caused by: unknown field `allowUnauthenticatedPermissions` for key `binds` at line 5 column 1

andreflosi, 2023-07-08:
**Describe the bug**
After the installation following the install.md at "https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install.md",  even after add a ip camera and use the test button, I cannot start the moonfire-nvr, checking the result of the command "nvr logs" commands, it show the below:

E20230708 12:45:11.130 main moonfire_nvr] Exiting due to error: unable to read /etc/moonfire-nvr.toml caused by: unknown field `allowUnauthenticatedPermissions` for key `binds` at line 5 column 1

**To Reproduce**
Steps to reproduce the behavior:
1. start the moonfire-nvr using the nvr start command;

2. checking the result of the command : sudo docker ps -a:
CONTAINER ID   IMAGE                           COMMAND                  CREATED         STATUS                                  PORTS     NAMES
84678d654004   scottlamb/moonfire-nvr:v0.7.5   "/usr/local/bin/moon…"   7 seconds ago   Restarting (1) Less than a second ago             moonfire-nvr
It showed in the status column restarting.

3. checking the result of the command nvr logs:
E20230708 12:44:42.783 main moonfire_nvr] Exiting due to error: unable to read /etc/moonfire-nvr.toml
caused by: unknown field `allowUnauthenticatedPermissions` for key `binds` at line 5 column 1

Backtrace:
   0: failure::backtrace::internal::InternalBacktrace::new
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/internal.rs:46:44
   1: failure::backtrace::Backtrace::new
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:121:35
      <failure::backtrace::Backtrace as core::default::Default>::default
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:140:13
   2: <failure::error::error_impl::ErrorImpl as core::convert::From<F>>::from
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/error_impl.rs:19:17
      <failure::error::Error as core::convert::From<F>>::from
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/mod.rs:36:18
   3: <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/core/src/result.rs:2064:27
      moonfire_nvr::cmds::run::read_config
             at src/server/src/cmds/run/mod.rs:133:18
      moonfire_nvr::cmds::run::run
             at src/server/src/cmds/run/mod.rs:138:18
   4: moonfire_nvr::Args::run
             at src/server/src/main.rs:68:29
      moonfire_nvr::main
             at src/server/src/main.rs:143:9
   5: core::ops::function::FnOnce::call_once
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/core/src/ops/function.rs:227:5
      std::sys_common::backtrace::__rust_begin_short_backtrace
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/sys_common/backtrace.rs:122:18
   6: std::rt::lang_start::{{closure}}
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/rt.rs:145:18
   7: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/core/src/ops/function.rs:259:13
      std::panicking::try::do_call
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panicking.rs:492:40
      std::panicking::try
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panicking.rs:456:19
      std::panic::catch_unwind
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panic.rs:137:14
      std::rt::lang_start_internal::{{closure}}
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/rt.rs:128:48
      std::panicking::try::do_call
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panicking.rs:492:40
      std::panicking::try
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panicking.rs:456:19
      std::panic::catch_unwind
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/panic.rs:137:14
      std::rt::lang_start_internal
             at /rustc/7737e0b5c4103216d6fd8cf941b7ab9bdbaace7c/library/std/src/rt.rs:128:20
   8: main
   9: __libc_start_main
  10: <unknown>

**Expected behavior**
I expected to open the webpage on the http://localhost:8080

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server (please complete the following information):**
  - If using Docker: `docker ps` + `docker images`
  
pi@raspberrypi:~/moonfire-nvr $ sudo docker ps --no-trunc
CONTAINER ID                                                       IMAGE                           COMMAND                             CREATED         STATUS                          PORTS     NAMES
dc9b1812888f7d42134602da62af720039de6bd25f6ded7a01d0b2f0df408f09   scottlamb/moonfire-nvr:v0.7.5   "/usr/local/bin/moonfire-nvr run"   8 minutes ago   Restarting (1) 22 seconds ago             moonfire-nvr

  - If building from git: `git describe --dirty` + `moonfire-nvr --version`

  - Attach a [log file](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#viewing-moonfire-nvrs-logs). Run with the `RUST_BACKTRACE=1` environment variable set if possible.

**Camera (please complete the following information):**
  - Camera manufacturer and model: Hikvision DS-2CD1023G0E
  - Firmware version: V5.7.12 build 220819

**Desktop (please complete the following information):**
 - OS: raspbian

**Additional context**

pi@raspberrypi:~/moonfire-nvr $ sudo cat /usr/local/bin/nvr
#!/bin/bash -e

# Set your timezone here.
tz="America/Sao_Paulo"

image_name="scottlamb/moonfire-nvr:v0.7.5"
container_name="moonfire-nvr"
common_docker_run_args=(
        --mount=type=bind,source=/var/lib/moonfire-nvr,destination=/var/lib/moonfire-nvr
        --mount=type=bind,source=/etc/moonfire-nvr.toml,destination=/etc/moonfire-nvr.toml

        # Add additional mount lines here for each sample file directory
        # outside of /var/lib/moonfire-nvr, e.g.:
        # --mount=type=bind,source=/media/nvr/sample,destination=/media/nvr/sample

        --user="$(id -u moonfire-nvr):$(id -g moonfire-nvr)"

        # This avoids errors with broken seccomp on older 32-bit hosts.
        # https://github.com/moby/moby/issues/40734
        --security-opt=seccomp:unconfined

        # This is the simplest way of configuring networking, although
        # you can use e.g. --publish=8080:8080 in the run) case below if you
        # prefer.
        --network=host

        # docker's default log driver won't rotate logs properly, and will throw
        # away logs when you destroy and recreate the container. Using journald
        # solves these problems.
        # https://docs.docker.com/config/containers/logging/configure/
        --log-driver=journald
        --log-opt="tag=moonfire-nvr"

        --env=RUST_BACKTRACE=1
        --env=TZ=":${tz}"
)

case "$1" in
run)
        shift
        exec docker run \
                --detach=true \
                --restart=unless-stopped \
                "${common_docker_run_args[@]}" \
                --name="${container_name}" \
                "${image_name}" \
                run \
                "$@"
        ;;
start|stop|logs|rm)
        exec docker "$@" "${container_name}"
        ;;
pull)
        exec docker pull "${image_name}"
        ;;
*)
        exec docker run \
                --interactive=true \
                --tty \
                --rm \
                "${common_docker_run_args[@]}" \
                "${image_name}" \
                "$@"
        ;;
esac

andreflosi, 2023-07-08:
the content of /etc/moonfire-nvr.toml

pi@raspberrypi:~/moonfire-nvr $ cat /etc/moonfire-nvr.toml
[[binds]]
ipv4 = "0.0.0.0:8080"
allowUnauthenticatedPermissions = { viewVideo = true }

[[binds]]
unix = "/var/lib/moonfire-nvr/sock"
ownUidIsPrivileged = true

IronOxidizer, 2023-07-08:
Removing the lines 
> allowUnauthenticatedPermissions

and

> ownUidIsPrivileged = true

fixed the issues for me, however, I now have a new issue where attempting live view from outside my local network results in the following browser error: `"may not load data from blob"`, not sure if it's related.

ldfsilva, 2023-07-08:
The initial error you got is because of discrepant `moonfire-nvr.toml` configuration between version `v0.7.5` and `v0.7.6`. 

Notice that `camelCase` was introduced for use in version `v0.7.6`, and that's a breaking change in comparison to previous versions (ie `v0.7.5` and previous), check out [`v0.7.6` release notes](https://github.com/scottlamb/moonfire-nvr/blob/master/CHANGELOG.md#076-2023-07-08) for details. 

I'd suggest you re-do your `moonfire-nvr.toml` with `camelCase` and set your `image_name` in `nvr` script to `v0.7.6` (ie: `image_name="scottlamb/moonfire-nvr:v0.7.6"`). Try that out and see if it'll then work as expected.

scottlamb, 2023-07-08:
> however, I now have a new issue where attempting live view from outside my local network results in the following browser error: "may not load data from blob", not sure if it's related.

I'm looking at that error in #277 . You didn't see that in `v0.7.5` and do in `v0.7.6`? That's an interesting data point.

andreflosi, 2023-07-10:
> The initial error you got is because of discrepant `moonfire-nvr.toml` configuration between version `v0.7.5` and `v0.7.6`.
>
> Notice that `camelCase` was introduced for use in version `v0.7.6`, and [...]

Hi, I did the suggested procedure, but it not worked.

 sudo head /usr/local/bin/nvr
#!/bin/bash -e

# Set your timezone here.
tz="America/Sao_Paulo"

image_name="scottlamb/moonfire-nvr:v0.7.6"
container_name="moonfire-nvr"
common_docker_run_args=(
        --mount=type=bind,source=/var/lib/moonfire-nvr,destination=/var/lib/moonfire-nvr
        --mount=type=bind,source=/etc/moonfire-nvr.toml,destination=/etc/moonfire-nvr.toml
--------------------------------------------------------------------------------------------------
cat /etc/moonfire-nvr.toml
[[binds]]
ipv4 = "0.0.0.0:8080"
allowUnauthenticatedPermissions = { viewVideo = true }

[[binds]]
unix = "/var/lib/moonfire-nvr/sock"
--------------------------------------------------------------------------------------------------
docker ps -a
CONTAINER ID   IMAGE                           COMMAND                  CREATED       STATUS                                  PORTS     NAMES
dc9b1812888f   scottlamb/moonfire-nvr:v0.7.5   "/usr/local/bin/moon…"   2 days ago    Restarting (1) Less than a second ago             moonfire-nvr

the procedure told by the [IronOxidizer], worked. At least the webserver ran and I could log in. Will test the liveview.

ldfsilva, 2023-07-11:
Your `docker ps -a` output indicates that your docker image is `v0.7.5`, it should be `v0.7.6` for the camel case arguments to work. You will need to either remove or rename your existing docker image.

Here's what I used to upgrade from `v0.7.5` to `v0.7.6`:

 - `sudo nvr stop`
 - `sudo docker rename moonfire-nvr moonfire-nvr-v0.7.5`
- Modified `/etc/moonfire-nvr.toml` config to conform with the new camelCase standard
- Modified `/usr/local/bin/nvr` to reflect the latest version `image_name="scottlamb/moonfire-nvr:v0.7.6"`
- `sudo nvr run`

closed 2023-10-21 by scottlamb