scottlamb/moonfire-nvr #286: Live view failed due to "may not load data from blob"

IronOxidizer, 2023-07-08:
**Describe the bug**
When attempting to use the live view, I get a black screen and the browser shows the error:
`Security Error: Content at https://nvr4.us.to/ may not load data from blob:https://mydomain.com/44abc5dc-750d-48d1-817d-2e6a52445592.`

This **only** happens on **Firefox** when I'm **outside of my local network**. While inside the network, or when using a chromium based browser, the live view works as intended.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Firefox
2. Go to `https://mydomain.com:8080`
3. Login
4. Go to live view and select a view
5. See error

**Expected behavior**
Live view appears

**Server (please complete the following information):**
  - `scottlamb/moonfire-nvr:v0.7.6`

**Desktop (please complete the following information):**
 - OS: Arch @ Linux 6.3.8, Ubuntu 20.04
 - Browser: Firefox 114, Firefox 115

**Additional context**
I understand this is probably not a moonfire issue, but is there a workaround that can be implemented in moonfire? Based on some quick research, the trigger might be paramaterized blobs, still not sure though.

scottlamb, 2023-07-09:
I see a similar error at https://github.com/scottlamb/moonfire-nvr/issues/277#issuecomment-1627480506 but haven't managed to figure out the fix.

But question here:

> This only happens on Firefox when I'm outside of my local network. While inside the network, or when using a chromium based browser, the live view works as intended.

How does your outside-the-network setup work? You have two different domain names as in that error message (`https://nvr4.us.to/` vs `https://mydomain.com/`)?

Also, am I correctly inferring from [this comment](https://github.com/scottlamb/moonfire-nvr/issues/285#issuecomment-1627452748) that it worked in v0.7.5? There have been a lot of server-side changes since then but really not much on the frontend side. The biggest thing was upgrading to React v18 but I don't see anything relevant in the release notes.

IronOxidizer, 2023-07-09:
`https://mydomain.com/` was just an example as to not leak my actual domain, but I guess I forgot anyways 😅

> Also, am I correctly inferring from https://github.com/scottlamb/moonfire-nvr/issues/285#issuecomment-1627452748 that it worked in v0.7.5?

Last working version of live-view for me was v0.7.2.

v0.7.3 introduced https://github.com/scottlamb/moonfire-nvr/issues/219 which broke live view, and wasn't fixed until v0.7.6 because of https://github.com/scottlamb/moonfire-nvr/issues/230. It's possible that this is new behavior that was introduced into Firefox in 15 months since v0.7.2

IronOxidizer, 2023-07-09:
Some more data points:

- Firefox 102 on Windows is broken regardless of local or external connection
- Firefox 115 on Windows is also broken regardless of local or external connection
- Chrome 114 on Windows is fully working

scottlamb, 2023-07-09:
Hmm. Skimmed Firefox release notes. Didn't notice anything relevant. I'm kind of stuck. :-(

IronOxidizer, 2023-07-09:
This seems like a Firefox bug based on:
- https://bugzilla.mozilla.org/show_bug.cgi?id=1628862
- https://bugzilla.mozilla.org/show_bug.cgi?id=1768052

There's also dozens of mentions of it on Github, with no root-cause: https://github.com/search?q=%22may+not+load+data+from+blob%22&type=issues

Though it's weird that this hasn't been an issue till now. @scottlamb Are you able to reproduce it on Firefox?

scottlamb, 2023-07-09:
Yes, I am able to reproduce. I also found some similar error messages on the web, but I'm not sure how relevant those things are, e.g. the first two you linked:

* https://bugzilla.mozilla.org/show_bug.cgi?id=1628862 sounds like it relates to a query string; we shouldn't be using one
* sounds like it has something to do with revoking the URL; we're not doing that AFAICT

At first I thought this was related to `Content-Security-Policy` but messing with that didn't help anything.

I might try firing up v0.7.2 with current Firefox and see if it's affected, and/or find some old Firefox versions to bisect.

scottlamb, 2023-07-09:
Okay, here's something:

* new Moonfire, Firefox 75 => same problem
* slightly older Moonfire (2667dd6), current Firefox => works fine

I guess I'll do a `git bisect` to narrow down what changed.

scottlamb, 2023-07-09:
Hmm, looks like it broke with 098b54c. Not sure why...

scottlamb, 2023-07-09:
...and specifically, updating to React 18 alone doesn't break it, but doing the [recommended](https://react.dev/blog/2022/03/08/react-18-upgrade-guide#updates-to-client-rendering-apis) switch from `ReactDOM.render` to `createRoot` does. Not seeing anything in their docs that looks relevant though.

scottlamb, 2023-09-30:
I finally distilled this down to a relatively minimal reproducible example and asked for help on stackoverflow. Fingers crossed someone will help me understand what's going on, and I can get it fixed.

* https://github.com/scottlamb/repro-blob-error
* https://stackoverflow.com/questions/77208669/why-does-my-mse-code-get-security-error-content-at-http-localhost3000-may

jlpoolen, 2023-10-14:
I just upgraded today, Oct 14, 2023, and encountered the same problem with Firefox 115.3.1esr (64-bit).  However, if I use Chrome Version 109.0.5414.120 (Official Build) (64-bit), I am able to view live "sub" feeds in the Live Views.  I loaded up the screen using the 3x3 and successfully displayed 9 cameras in Chrome.

commit 8b5f2b4b (2024-04-17): work on Firefox!

Fixes #286.

I dug into Firefox code a bit to understand this but got a lost. But
I guess there are different policies for what's accessible via
`video.src = ""` than for `<video src="">`. This works for now, matches
what other players such as `mpegts.js` do, and is closer to what Safari
MME (required on iPhone) needs anyway. (With MME, apparently you have to
use `video.srcObject`, or the `MediaSource`'s `sourceopen` event will
never fire.)
closed 2024-04-17 by scottlamb