scottlamb/moonfire-nvr #289: Unable to add a sample directory

Anyn0ne, 2023-07-22:
**Describe the bug**
`nvr config` crashes when adding a sample directory, whitout any error message.

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo nvr config 2>debug-log`
2. choose _Directories and retention_ and then _new sample file dir_
3. specify the folder (/var/lib/moonfire-nvr/sample in my case)
4. you won't see any error, nothing happens. nvr config crashes "silently"

**Expected behavior**
The folder should be added in order to use it when adding a camera

**Screenshots**
not a screenshot, but here is what I get with `journalctl -xe` :
```
Jul 22 10:44:41 DietPi moonfire-nvr[1180]: [3B blob data]
Jul 22 10:44:41 DietPi moonfire-nvr[1180]: [3B blob data]
Jul 22 10:44:41 DietPi moonfire-nvr[1180]: [3B blob data]
Jul 22 10:44:41 DietPi moonfire-nvr[1180]: [3B blob data]
...
Jul 22 10:44:52 DietPi moonfire-nvr[1180]: [80B blob data]
Jul 22 10:44:52 DietPi moonfire-nvr[1180]: [704B blob data]
Jul 22 10:45:07 DietPi moonfire-nvr[1180]: [1.3K blob data]
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./src/server/base/tracing_setup.rs:105:22
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    1: core::ops::function::Fn::call
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/ops/function.rs:79:5
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    2: <alloc::boxed::Box<F,A> as core::ops::function::Fn<Args>>::call
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/alloc/src/boxed.rs:1987:9
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    3: std::panicking::rust_panic_with_hook
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/std/src/panicking.rs:695:13
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    4: std::panicking::begin_panic_handler::{{closure}}
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/std/src/panicking.rs:580:13
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    5: std::sys_common::backtrace::__rust_end_short_backtrace
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/std/src/sys_common/backtrace.rs:150:18
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    6: rust_begin_unwind
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/std/src/panicking.rs:578:5
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    7: core::panicking::panic_fmt
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/panicking.rs:67:14
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    8: core::panicking::panic
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/panicking.rs:117:5
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:    9: core::option::Option<T>::unwrap
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/option.rs:950:21
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   10: moonfire_nvr::cmds::config::dirs::add_dir_dialog::{{closure}}
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./src/server/src/cmds/config/dirs.rs:270:22
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   11: cursive_core::event::Callback::from_fn::{{closure}}
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/event.rs:189:13
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   12: cursive_core::cursive::Cursive::on_event
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/cursive.rs:840:17
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   13: cursive_core::cursive_run::CursiveRunner<C>::process_events
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/cursive_run.rs:125:13
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   14: cursive_core::cursive_run::CursiveRunner<C>::step
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/cursive_run.rs:216:34
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   15: cursive_core::cursive_run::CursiveRunner<C>::run
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/cursive_run.rs:239:13
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   16: cursive_core::cursive::Cursive::try_run_with
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive_core-0.3.6/src/cursive.rs:922:9
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   17: cursive::cursive_runnable::CursiveRunnable::try_run
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive-0.20.0/src/cursive_runnable.rs:80:9
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   18: cursive::cursive_runnable::CursiveRunnable::run
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./.cargo/registry/src/index.crates.io-6f17d22bba15001f/cursive-0.20.0/src/cursive_runnable.rs:75:9
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   19: moonfire_nvr::cmds::config::run
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./src/server/src/cmds/config/mod.rs:58:5
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   20: moonfire_nvr::Args::run
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./src/server/src/main.rs:43:32
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   21: moonfire_nvr::main
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at ./src/server/src/main.rs:94:11
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   22: core::ops::function::FnOnce::call_once
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/ops/function.rs:250:5
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   23: std::sys_common::backtrace::__rust_begin_short_backtrace
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:              at /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/std/src/sys_common/backtrace.rs:134:18
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   24: main
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   25: __libc_start_main
Jul 22 10:45:07 DietPi moonfire-nvr[1180]:   26: <unknown>
Jul 22 10:45:07 DietPi moonfire-nvr[1180]: 
Jul 22 10:45:07 DietPi systemd[1]: docker-968b65a8242fe54bf213fb209f2ab60008bfb5fb7ea9c097ce7f61234791f9ad.scope: Deactivated successfully.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit docker-968b65a8242fe54bf213fb209f2ab60008bfb5fb7ea9c097ce7f61234791f9ad.scope has successfully entered the 'dead' state.
Jul 22 10:45:07 DietPi moonfire-nvr[1180]: [60B blob data]
Jul 22 10:45:07 DietPi dockerd[1180]: time="2023-07-22T10:45:07.819010462+02:00" level=info msg="ignoring event" container=968b65a8242fe54bf213fb209f2ab60008bfb5fb7ea9c097ce7f61234791f9ad module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Jul 22 10:45:07 DietPi containerd[512]: time="2023-07-22T10:45:07.819866351+02:00" level=info msg="shim disconnected" id=968b65a8242fe54bf213fb209f2ab60008bfb5fb7ea9c097ce7f61234791f9ad
Jul 22 10:45:07 DietPi containerd[512]: time="2023-07-22T10:45:07.820121721+02:00" level=warning msg="cleaning up after shim disconnected" id=968b65a8242fe54bf213fb209f2ab60008bfb5fb7ea9c097ce7f61234791f9ad namespace=moby
Jul 22 10:45:07 DietPi containerd[512]: time="2023-07-22T10:45:07.820285888+02:00" level=info msg="cleaning up dead shim"
Jul 22 10:45:07 DietPi containerd[512]: time="2023-07-22T10:45:07.853265536+02:00" level=warning msg="cleanup warnings time=\"2023-07-22T10:45:07+02:00\" level=info msg=\"starting signal loop\" namespace=moby pid=1768 runtime=io.containerd.runc.v2\n"
Jul 22 10:45:07 DietPi systemd[1]: var-lib-docker-overlay2-1b913a2a6ec6e2ca36f45e84ef9a1fc016a25fa6ffeb540b974022117eb06cca-merged.mount: Deactivated successfully.
```

**Server (please complete the following information):**
During configuration step, moonfire-nvr should not be running, so docker ps gives me nothing.

`docker images` is returning

```
REPOSITORY               TAG       IMAGE ID       CREATED        SIZE
rss-ffmpeg               latest    700354def66e   9 days ago     745MB
scottlamb/moonfire-nvr   v0.7.6    c7449ebdd919   13 days ago    297MB
bluenviron/mediamtx      latest    e058be0e827d   2 weeks ago    25.5MB
hello-world              latest    b038788ddb22   2 months ago   9.14kB
```

**Additional context**
The whole system is running on a raspberry pi 4. The OS used is dietpi

ORIOLESFan02, 2023-07-31:
I have the same problem. Fedora Server 38. However `sudo nvr config 2>debug-log` doesn't work for me, but `sudo nvr config` does.

For mine this is what mine says when "crashes" Also using `/var/lib/moonfire-nvr/sample`
```
2023-07-31T18:49:35.445971  INFO main moonfire_nvr::cmds: Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.31.1
2023-07-31T18:49:35.573376  INFO main moonfire_db::db: Loading video sample entries
2023-07-31T18:49:35.573527  INFO main moonfire_db::db: Loaded 0 video sample entries
2023-07-31T18:49:35.573568  INFO main moonfire_db::db: Loading sample file dirs
2023-07-31T18:49:35.573728  INFO main moonfire_db::db: Loaded 0 sample file dirs
2023-07-31T18:49:35.573762  INFO main moonfire_db::db: Loading cameras
2023-07-31T18:49:35.573868  INFO main moonfire_db::db: Loaded 0 cameras
2023-07-31T18:49:35.573892  INFO main moonfire_db::db: Loading streams
2023-07-31T18:49:35.574281  INFO main moonfire_db::db: Loaded 0 streams

scottlamb, 2023-08-03:
Thanks for the clear report. I can reproduce both problems mentioned here:

* the crash itself, which seems to be something related to 3d40a39b93ee42adf3c8fd835a08e627d2cc0c69 (@sky1e). I haven't looked too closely yet but it seems to have different behavior with pressing enter within the field (works) vs tabbing over and pressing enter on the `<Add>` button (crashes).
* `2>debug-log` not producing output in `debug-log`; ebcdd7608427addb4f1a51910f1ee39642c57628 stopped using stderr, and the instructions didn't get updated...

closed 2023-08-03 by scottlamb

scottlamb, 2023-08-03:
Fix released in [v0.7.7](https://github.com/scottlamb/moonfire-nvr/releases/tag/v0.7.7).

commit 02ac1a55 (2023-08-03): fix #289: crash on sample dir `Add` button press