scottlamb/moonfire-nvr #295: Max video duration hour options does not display different video length segments

ldfsilva, 2023-11-25:
**Describe the bug**
The max video duration hour options (1, 4, 24, infinite) does not display different video length segments in the playback UI, despite the selection the video segments remain the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Moonfire UI'
2. Click on 'Login'
3. Select either 'Main' or 'Sub' streams
4. Select a date that have recordings
5. Choose the 'Max video duration' of choice (either of 1, 4, 24 or infinite hours)
6. See the error, despite the selection the displayed 'start' and 'end' video segment on the right panel remains the same

**Expected behavior**
The video segments displayed to reflect the chosen 'Max video duration' length accordingly, either of the hour options offered (1, 4, 24 or infinite).

Note in screenshot example ([screenshots/list.png](https://github.com/scottlamb/moonfire-nvr/blob/master/screenshots/list.png)) shows it used to work, given the selection is set to 1 hours and the video segments have 1 hour duration.

**Screenshots**
See below screenshots for 1 and 4 hours selection, note that despite selection the video segments are all 5 minutes apart from each other.

1 hour
[image]

4 hours
[image]

**Server (please complete the following information):**
  - `moonfire-nvr --version`

```
[moonfire-nvr@nuc sample]$ moonfire-nvr --version
Version: v0.7.8
```

**Camera (please complete the following information):**
  - Amcrest IP2M-841

**Desktop (please complete the following information):**
 - Browser [e.g. chrome, safari]; 
   - Firefox 117.0 (64-bit)
   - Chrome 119.0.6045.159 (Official Build) (x86_64)

**Additional context**
The behavior is the same if I try from mobile (Chrome and DuckDuckGo browsers), so it doesn't seem to be a browser issue. Here's 100 most recent `recordings` entries from the following, can provide additional data if that'd help better understand state of recordings.

```
sudo -u moonfire-nvr moonfire-nvr sql "select * from recording order by start_time_90k desc limit 100" 
```
https://gist.github.com/ldfsilva/4bb33f4d02f9f0e009ca7c16bb87a60f

ldfsilva, 2023-11-26:
An example exposing the duration with this patch https://github.com/scottlamb/moonfire-nvr/compare/master...ldfsilva:moonfire-nvr:add_duration, I wonder if this is led by non-contiguous video samples, which upon closer look to [screenshots/list.png](https://github.com/scottlamb/moonfire-nvr/blob/master/screenshots/list.png) also had same behavior (eg; 12:48:31 - 12:53:25).

[image]

scottlamb, 2026-02-17:
I think what was happening here was some error caused the video stream to terminate and restart every 5 minutes, so the 1 hour / 4 hour / ... limit never kicked in. I've since made a change to the UI that actually shows when the stream ends due to an error via an exclamation mark and a mouseover:

[image]

so I think that fixes it in the sense that the UI is now less confusing when this happens. We may or may not have also since fixed whatever stream error was happening; if not feel free to open another issue with logs.

closed 2026-02-17 by scottlamb