scottlamb/moonfire-nvr #296: Issue installing in docker

deanfourie1, 2023-12-03:
**Describe the bug**
Unable to start docker container.

Also, when atempting to edit moonfire-nvr.toml

`sudo nano /etc/moonfire-nvr.toml`
`[ "/etc/moonfire-nvr.toml" is a directory ]`

I cannot edit the file as it says its a directory. The documentary clearly states to edit the config file @ /etc/moonfire-nvr.toml.

Error:
Error response from daemon: unable to find user UID: no matching entries in passwd file

**To Reproduce**
Steps to reproduce the behavior:
1. Install Docker
2. Run docker container from docker-compose.yml file
3. docker compose up

**Expected behavior**
Container to run.

**Screenshots**

**Server (please complete the following information):**
  - `0.7.1`
  - Cannot add log.
  - 
**Camera (please complete the following information):**
  - Camera manufacturer and model: [e.g. Reolink RLC-410]
  - Firmware version: [e.g. `V2.0.0.1215_16091800`]

**Desktop (please complete the following information):**
 - OS: Debian 12
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

scottlamb, 2023-12-05:
Just wanted to say: sorry if my terse response above came off as rude. For context, I'm struggling to find time to work on Moonfire, and I was reluctant to even have a Docker option in the install instructions for fear of confusion (see [here](https://github.com/scottlamb/moonfire-nvr/discussions/294#discussioncomment-7646729)). I'm a bit frustrated to see that seems to be the case.

If you have any ideas to improve the instructions, PRs are welcome...

fakuivan, 2023-12-05:
As @scottlamb mentioned, you need to first create the config file you're trying to pass onto the container. The default behaviour for docker is to:
1. Look for the file on the host file system
2. If found, mount it, being a directory or a file
3. If not found, mount it as a directory

If the file `/etc/moonfire-nvr.toml` does not exist on the host, then docker will create a directory with that name, despite it having the `.toml` extension. As far as I can see this has been adressed in the docs so I think we're ok.