scottlamb/moonfire-nvr #298: stream test failed

maomj200811, 2024-01-09:
**Describe the bug**
A clear and concise description of the problem.

the following question is : DESCRIBE Cseq=1: Describe response not of expected application/sdp content type: Response

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server (please complete the following information):**
  - `moonfire-nvr --version`
  - Attach a [log file](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#viewing-moonfire-nvrs-logs). Run with the `RUST_BACKTRACE=1` environment variable set if possible.

**Camera (please complete the following information):**
  - Camera manufacturer and model: [e.g. Reolink RLC-410]
  - Firmware version: [e.g. `V2.0.0.1215_16091800`]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

maomj200811, 2024-01-09:
version v0.7.11

scottlamb, 2024-01-09:
What camera manufacturer/model/version is this?

The response is supposed to include `Content-Type: application/sdp` and doesn't but otherwise looks plausible. I can probably just have Retina (Moonfire's protocol layer) assume that content type if none is specified.

maomj200811, 2024-01-09:
but this video can play normally using vlc player, video source is published using ffmpeg

scottlamb, 2024-01-09:
VLC is pretty forgiving. Moonfire can be too; I'll release an updated version shortly that should work better.

scottlamb, 2024-01-09:
v0.7.12 *should* be out shortly...but Github Actions seems to be in a bad mood, so we'll see.

maomj200811, 2024-01-09:
> VLC is pretty forgiving. Moonfire can be too; I'll release an updated version shortly that should work better.

what time it will updated newest version?

scottlamb, 2024-01-09:
When [this action](https://github.com/scottlamb/moonfire-nvr/actions/runs/7457036703) completes...so right now! Please give it a try.

maomj200811, 2024-01-09:
thanks. a lot it's ok now..  but now I has new questions  about moonfire-nvr, my camera record storage is zero size. the following.

[image]

scottlamb, 2024-01-09:
Yeah, that's definitely not right. There are likely clues in the logs.

scottlamb, 2024-01-09:
I see you opened #300 for that; closing this issue out.

closed 2024-01-09 by scottlamb