scottlamb/moonfire-nvr #302: Camera feed full of disconnects and short segments

a-bali, 2024-01-13:
I'm trying to run Moonfire NVR with a cheap Chinese wifi security camera (not sure if it has any meaningful brand or version, comes with the ICSee mobile app). The camera which works fine with ffmpeg via TCP/RTSP, producing continuous 1-hour segments without problems with the following command:
`ffmpeg -rtsp_transport tcp -i rtsp://ip:554/user=..._password=..._channel=1_stream=0.sdp?real_stream -an -vcodec copy -f segment -segment_time 600 -strftime 1 -loglevel warning ...`

With latest (0.7.12) Moonfire NVR (same URL as above, TCP transport) however, the feed is full of random disconnects (sometimes after just a few seconds, sometimes after minutes, but sometimes only after a couple of hours - see screenshot below), therefore the list of segments is very fragmented and of course it is also not possible to watch an arbitrary timespan at once with these small fragments.

Would it be possible to somehow improve the feed continuity as ffmpeg does (I don't get any disconnects there) or merge the segments in the background and thus hide the disconnects from the user?

Let me know if I can somehow provide further data for debugging.

[image]

scottlamb, 2024-01-13:
Could you attach some logs? They should explain each disconnection.

a-bali, 2024-01-14:
I don't seem to have anything specific in the logs, apart from entries like below which confirm that the segment added is just a couple of seconds. Is it possible to increase the log level somehow?

```
2024-01-14T07:16:09.362783  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1261 reason="0 sec after start of 1 minute nyaralo-main recording 1/2352"}: moonfire_db::db: fl
ush complete:
/home/bali/nvr/main: added 3M 291K 843B in 1 recordings (1/2352), deleted 0B in 0 (), GCed 0 recordings ().
2024-01-14T07:17:09.386288  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1262 reason="0 sec after start of 1 minute nyaralo-main recording 1/2353"}: moonfire_db::db: fl
ush complete:
/home/bali/nvr/main: added 3M 22K 91B in 1 recordings (1/2353), deleted 0B in 0 (), GCed 0 recordings ().
2024-01-14T07:17:39.391519  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1263 reason="0 sec after start of 30 seconds nyaralo-main recording 1/2354"}: moonfire_db::db:
flush complete:
/home/bali/nvr/main: added 1M 525K 355B in 1 recordings (1/2354), deleted 0B in 0 (), GCed 0 recordings ().
2024-01-14T07:17:49.369261  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1264 reason="0 sec after start of 10 seconds nyaralo-main recording 1/2355"}: moonfire_db::db:
flush complete:
/home/bali/nvr/main: added 487K 497B in 1 recordings (1/2355), deleted 0B in 0 (), GCed 0 recordings ().
2024-01-14T07:18:09.357768  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1265 reason="0 sec after start of 20 seconds nyaralo-main recording 1/2356"}: moonfire_db::db:
flush complete:
/home/bali/nvr/main: added 900K 900B in 1 recordings (1/2356), deleted 0B in 0 (), GCed 0 recordings ().
```

a-bali, 2024-01-15:
Also these "parameter changes":

```
2024-01-15T06:15:34.885377 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: retina::codec::h264: sps: SeqParameterSet {
    profile_idc: ProfileIdc(
        77,
    ),
    constraint_flags: ConstraintFlags {
        flag0: false,
        flag1: false,
        flag2: false,
        flag3: false,
        flag4: false,
        flag5: false,
        reserved_zero_two_bits: 0,
    },
    level_idc: 41,
    seq_parameter_set_id: ParamSetId(
        0,
    ),
    chroma_info: ChromaInfo {
        chroma_format: YUV420,
        separate_colour_plane_flag: false,
        bit_depth_luma_minus8: 0,
        bit_depth_chroma_minus8: 0,
        qpprime_y_zero_transform_bypass_flag: false,
        scaling_matrix: SeqScalingMatrix,
    },
    log2_max_frame_num_minus4: 12,
    pic_order_cnt: TypeTwo,
    max_num_ref_frames: 2,
    gaps_in_frame_num_value_allowed_flag: true,
    pic_width_in_mbs_minus1: 119,
    pic_height_in_map_units_minus1: 67,
    frame_mbs_flags: Frames,
    direct_8x8_inference_flag: true,
    frame_cropping: Some(
        FrameCropping {
            left_offset: 0,
            right_offset: 0,
            top_offset: 0,
            bottom_offset: 4,
        },
    ),
    vui_parameters: Some(
        VuiParameters {
            aspect_ratio_info: None,
            overscan_appropriate: Unspecified,                                                                                                                                                                      video_signal_type: None,
            chroma_loc_info: None,
            timing_info: Some(
                TimingInfo {
                    num_units_in_tick: 1,
                    time_scale: 24,
                    fixed_frame_rate_flag: true,
                },
            ),
            nal_hrd_parameters: None,
            vcl_hrd_parameters: None,
            low_delay_hrd_flag: None,
            pic_struct_present_flag: false,
            bitstream_restrictions: None,
        },
    ),
}
2024-01-15T06:15:34.886011 DEBUG       s-nyaralo-main streamer{stream="nyaralo-main"}: moonfire_nvr::stream: nyaralo-main: parameter change:
old: VideoSampleEntryToInsert { data: Length: 131 (0x83) bytes
0000:   00 00 00 83  61 76 63 31  00 00 00 00  00 00 00 01   ....avc1........
0010:   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0020:   07 80 04 38  00 48 00 00  00 48 00 00  00 00 00 00   ...8.H...H......
0030:   00 01 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0040:   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0050:   00 00 00 18  ff ff 00 00  00 2d 61 76  63 43 01 4d   .........-avcC.M
0060:   00 29 ff e1  00 15 27 4d  00 29 8d 6e  07 80 22 7e   .)....'M.).n.."~
0070:   58 40 00 00  03 00 40 00  00 06 21 01  00 05 28 ee   X@....@...!...(.
0080:   02 1c 80                                             ..., rfc6381_codec: "avc1.4d0029", width: 1920, height: 1080, pasp_h_spacing: 1, pasp_v_spacing: 1 }
new: VideoSampleEntryToInsert { data: Length: 130 (0x82) bytes
0000:   00 00 00 82  61 76 63 31  00 00 00 00  00 00 00 01   ....avc1........
0010:   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0020:   07 80 04 38  00 48 00 00  00 48 00 00  00 00 00 00   ...8.H...H......
0030:   00 01 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0040:   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ................
0050:   00 00 00 18  ff ff 00 00  00 2c 61 76  63 43 01 4d   .........,avcC.M
0060:   00 29 ff e1  00 15 27 4d  00 29 8d 6e  07 80 22 7e   .)....'M.).n.."~
0070:   58 40 00 00  03 00 40 00  00 06 21 01  00 04 28 ee   X@....@...!...(.
0080:   05 72                                                .r, rfc6381_codec: "avc1.4d0029", width: 1920, height: 1080, pasp_h_spacing: 1, pasp_v_spacing: 1 }
2024-01-15T06:15:34.891664 DEBUG               sync-1 syncer{path=/home/bali/nvr/main}: moonfire_db::writer: 1: have remaining quota of -10882781184
2024-01-15T06:15:34.897675  INFO               sync-1 syncer{path=/home/bali/nvr/main}:flush{flush_count=1580 reason="0 sec after start of 19 seconds nyaralo-main recording 1/4005"}: moonfire_db::db:
flush complete:
/home/bali/nvr/main: added 195K 163B in 1 recordings (1/4005), deleted 0B in 0 (), GCed 0 recordings ().
```

scottlamb, 2024-01-15:
> caused by: RTSP framing error: Invalid RTSP message; buffered:

This one is unfortunate. I've seen something similar with cameras using the live555 library. When the TCP window fills (maybe as a result of wifi congestion or something), they get a partial `write`—the kernel write what it could to the buffer and returned to userspace. They don't handle this well. The correct thing to do would be to monitor for write availability, then when the kernel indicates that, repeat the `write` with only the remaining portion. But poorly written cameras do all manner of dumb things instead—repeating the whole write, skipping the remaining bytes, blocking the whole thread (that's supposed to be doing other things) until there's availability, dropping the connection right away. The first two break the protocol, producing this error.

If you got a packet capture with e.g. Wireshark it'd probably confirm that the TCP window filled shortly before this happened.

So why does ffmpeg keep the connection? Basically I think they skip forward in the incoming stream until they find something that looks like a properly framed message. The preceding message is likely corrupt; and some stuff gets skipped; but they keep the connection open where Retina/Moonfire currently drops it. I'm not thrilled about that approach but it is possible to support doing the same in Retina. 🤷 

You could try using UDP. Some stuff may still get dropped (because RTP-over-UDP doesn't have any retry logic at all) but at least there wouldn't be any corrupt RTP messages. There's still an RTSP TCP connection that in theory can be affected by the same bug, but it's only used for keepalives after the session is started, so it's unlikely that the TCP window will ever fill.

> Also these "parameter changes":

I'm not sure what the actual change was here; it must have been pretty subtle. I'll dig into those bytes later. But it's probably not too hard to make these ones appear as one entry and play without break.

> Would it be possible to ... merge the segments in the background and thus hide the disconnects from the user?

fwiw, my ideal UI would be totally different, based on a scrub bar sort of thing. You select when to start playing, and it would likely jump to the next segment any time the connection was lost like this. But it's a lot of effort to implement and I don't know when I'll find the time for it... :-(

scottlamb, 2024-01-15:
The parameter change is the following...

```diff
--- before.debug	2024-01-15 13:40:20
+++ after.debug	2024-01-15 13:40:26
@@ -78,7 +78,7 @@
             num_ref_idx_l1_default_active_minus1: 0,
             weighted_pred_flag: false,
             weighted_bipred_idc: 0,
-            pic_init_qp_minus26: 8,
+            pic_init_qp_minus26: 5,
             pic_init_qs_minus26: 0,
             chroma_qp_index_offset: 0,
             deblocking_filter_control_present_flag: true,
```

...which is a detail of how the frames are encoded. I've never seen a camera change this sort of parameter mid-stream on its own, but it seems like a reasonable thing to do, and I don't think it will be a big change for Moonfire to make it seamless.

a-bali, 2024-01-16:
Thanks for the detailed investigation!

> You could try using UDP. Some stuff may still get dropped (because RTP-over-UDP doesn't have any retry logic at all) but at least there wouldn't be any corrupt RTP messages. There's still an RTSP [...]

Strange thing is that the camera works with UDP using ffmpeg (with `-rtsp_transport udp`), but not with Moonfire, see below. I'm not sure I understand the client/server ports in the Retina client setup, maybe this could be the reason for getting a connection refused error?

```
2024-01-16T08:32:06.409783  INFO s-nyaralo-main streamer{stream="nyaralo-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.33.200:554/user=XXX_password=XXX_channel=1_stream=0.sdp?real_stream

        0,
    ),
    chroma_info: ChromaInfo {
        chroma_format: YUV420,
        separate_colour_plane_flag: false,
        bit_depth_luma_minus8: 0,
        bit_depth_chroma_minus8: 0,
        qpprime_y_zero_transform_bypass_flag: false,
        scaling_matrix: SeqScalingMatrix,
    },
    log2_max_frame_num_minus4: 12,
    pic_order_cnt: TypeTwo,
    max_num_ref_frames: 2,
    gaps_in_frame_num_value_allowed_flag: true,
    pic_width_in_mbs_minus1: 119,
    pic_height_in_map_units_minus1: 67,
    frame_mbs_flags: Frames,
    direct_8x8_inference_flag: true,
    frame_cropping: Some(
        FrameCropping {
            left_offset: 0,
            right_offset: 0,
            top_offset: 0,
            bottom_offset: 4,
        },
    ),
    vui_parameters: Some(
        VuiParameters {
            aspect_ratio_info: None,
            overscan_appropriate: Unspecified,
            video_signal_type: None,
            chroma_loc_info: None,
            timing_info: Some(
                TimingInfo {
                    num_units_in_tick: 1,
                    time_scale: 24,
                    fixed_frame_rate_flag: true,
                },
            ),
            nal_hrd_parameters: None,
            vcl_hrd_parameters: None,
            low_delay_hrd_flag: None,
            pic_struct_present_flag: false,
            bitstream_restrictions: None,
        },
    ),
}
2024-01-16T08:32:06.422046 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: moonfire_nvr::stream: connected to "nyaralo-main", tool None
2024-01-16T08:32:06.427634 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: retina::client: SETUP response: Response {
    version: V1_0,
    status: Ok,
    reason_phrase: "OK",
    headers: Headers(
        {
            HeaderName(
                "Cache-Control",
            ): HeaderValue(
                "private",
            ),
            HeaderName(
                "Cseq",
            ): HeaderValue(
                "2",
            ),
            HeaderName(
                "Server",
            ): HeaderValue(
                "H264DVR 1.0",
            ),
            HeaderName(
                "Session",
            ): HeaderValue(
                "18426930;timeout=60",
            ),
            HeaderName(
                "Transport",
            ): HeaderValue(
                "RTP/AVP;unicast;mode=PLAY;source=192.168.33.200;client_port=20566-20567;server_port=40004-40005;ssrc=0",
            ),
            HeaderName(
                "x-Dynamic-Rate",
            ): HeaderValue(
                "1",
            ),
        },
    ),
    body: b"",
}
2024-01-16T08:32:06.427782 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: retina::client: established session "18426930", timeout=60s
2024-01-16T08:32:06.474171 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: retina::client: no PLAY seq on stream 0
2024-01-16T08:32:06.560364 DEBUG tokio-runtime-worker streamer{stream="nyaralo-main"}: retina::client: Ignoring UDP connection refused error

```

scottlamb, 2024-02-13:
I'll roll a new release soon-ish (hopefully tonight) with the commit above that should seamlessly handle those `parameter change` events, rather than causing separate rows in the UI. I don't have a camera that does this though so I'll need your feedback on if it's working as well as I hope!

As for the UDP stuff, the "connection refused" stuff should be totally unimportant. (When I get around to it, I'll alter the message to make this clearer.) Is there any following log message that might explain what's going on? I might need a packet capture (e.g. with Wireshark) to understand this if not.

scottlamb, 2024-02-13:
> I don't have a camera that does this though so I'll need your feedback on if it's working as well as I hope!

Hmm, actually, on one of my cameras, if I change the frame rate in the UI, it alters the video sample entry without ending the stream. This isn't frequent/unprompted as with your camera, but it allows me to test my change. Seems to work!

scottlamb, 2024-02-13:
> I'll roll a new release soon-ish (hopefully tonight) with the commit above that should seamlessly handle those `parameter change` events, rather than causing separate rows in the UI. 

https://github.com/scottlamb/moonfire-nvr/releases/tag/v0.7.13

scottlamb, 2024-04-18:
Did this have a significant effect? The remaining short streams should be actual disconnects. I also have been meaning to add a mouseover to the UI that shows the reason for a disconnect rather than requiring you go to the logs.

commit 0422593e (2024-06-01): ui list view: tool tip to see why recording ended

Users are often puzzled why there are short recordings. Previously
the only way to see this was to examine Moonfire's logs. This should
be a much better experience to find it right in the UI where you're
wondering, and without the potential the logs are gone.

Fixes #302
closed 2024-06-01 by scottlamb