scottlamb/moonfire-nvr #306: Mention docker image in READMNE

ppbrown, 2024-01-31:
Please update readme to directly point to  whereever is the new location for https://hub.docker.com/r/scottlamb/moonfire-nvr ?

for that matter, it would be nice if you updated the thing THERE, to directly point to the new version, instead of just vaguely mentioning "ghcr.io"

scottlamb, 2024-01-31:
What are you hoping to learn from that link that isn't present in the linked install instructions?

ppbrown, 2024-01-31:
Just a faster absorption.
Most github projects that have an OFFICIAL docker image, point to it from the top level README.
And then they have docker-specific install instructions, at the top level of the dockerhub page.

Nice example of the latter:

https://hub.docker.com/r/zoneminderhq/zoneminder

scottlamb, 2024-01-31:
ghcr.io doesn't let you customize its package pages at all AFAICT; it just shows the github project's README, so linking to it from the github project's README would be circular and unhelpful. The package is [here](https://github.com/scottlamb/moonfire-nvr/pkgs/container/moonfire-nvr) but there's really nothing to see.

I'd rather not follow that zoneminder page's example. It tells me how to start a Docker image but that's not really enough to set up and use the software. It says "Note: Detailled usage instructions for the development and release Dockerfiles are contained within each Dockerfile.", but looking through the dockerfiles is really not the right thing to expect non-devs to do, and when I looked at [one of the several Dockerfiles](https://github.com/ZoneMinder/zmdockerfiles/blob/master/release/ubuntu20.04/Dockerfile), it didn't describe how to set up and use the software either. I would have expected a link to the software's official [installation guide](https://zoneminder.readthedocs.io/en/latest/installationguide/index.html), except that page has a banner at the top saying "This entire guide needs to be checked/updated as needed", so I assume they just don't have any decent documentation.

https://hub.docker.com/r/scottlamb/moonfire-nvr links to an installation guide that I expect to actually work. Docker isn't my recommended way to install, but Docker installation is described on that page.

ppbrown, 2024-01-31:
I have to run it  in docker. theres too many conflicts with other things running on the machine.

Perhaps you might split your main install instructions, into separate pages for normal, vs docker-specific things.
Then pointing the dockerhub page, at the docker specific install instructions, would be very helpful.

You say in the main readme, that you want to "make it great!"

51% of making that happen, is making the documentation easy to read, and easy to use.
Currently, from a new user perspective, it is neither. 

You're writing it like a coder who just wants to get back to coding :-) rather than a docs writer who wants new users to have as painless an experience as possible.

Speaking as a software writer myself, trying to switch hats from coder to tech writer, has other benefits.
if you can change gears to look at things from a new user perspective, sometimes writing the new-user docs, makes you realize, "Huh... This setup procedure could use some cleanup, too"

scottlamb, 2024-01-31:
I've put a lot of effort into good user-facing docs. PRs that improve them are welcome. Drive-by complaints are not.

closed 2024-01-31 by scottlamb

ppbrown, 2024-01-31:
Telling a new user, who is giving you feedback "your docs for new users are lacking",

"hey, make them better yourself".

is the epitome of Bad user interaction. A person has to UNDERSTAND THE PRODUCT, before they can write better docs.
I've already said I'm a new user, and dont understand the product.

Poor attitude, man.
I'll look elsewhere for my nvr needs.

scottlamb, 2024-01-31:
> I'll look elsewhere for my nvr needs.

Sounds good.

There are a lot of ways I'd like to make Moonfire NVR easier to install and use, like having all the configuration possible through the web UI rather than having to start up a separate TUI-based config interface, shut that down, and then go to the web interface. But I have very little cycles to spend on the project, and arguing with you on github issues isn't helping.