scottlamb/moonfire-nvr #307: How can I convert the recordings to an mp4 format?

jonathan84clark, 2024-02-13:
**Describe the bug**
This isn't a bug. I was wondering if you could help point me to some code that would allow me to convert the recording files into a standard .mp4 format? 

I've been looking through the code and it's quite extensive. I haven't yet been able to find anything that can do this. Eventually I want to build a VS app (probably in C#) that can do this.

Anyway, if you can help me with this, that would be great!

Thanks,

Jonathan L Clark
Software Engineer
Special Services Group LLC

scottlamb, 2024-02-13:
I think of Moonfire's server as a DBMS specialized in video. When you use a DBMS like PostgreSQL or Oracle or whatever, you wouldn't try to write your own code to convert their internal storage to some standard representation. You'd talk to their server via their protocol. Similarly, I consider all of Moonfire's storage (SQLite database and the sample file directory) to be a black box. You get what you want out through its HTTP API. And if the HTTP API doesn't provide what you need, file a feature request to extend it.

So my answer to your question is: use the [`GET /api/cameras/<uuid>/<stream>/view.mp4
` endpoint](https://github.com/scottlamb/moonfire-nvr/blob/master/ref/api.md#get-apicamerasuuidstreamviewmp4).

jonathan84clark, 2024-02-13:
Thanks for getting back to me so quickly. I'm still trying to wrap my head around moonfire. I haven't worked with a tool quite like this before (it's really cool by the way!). I need to think through this a little; determine if this solution will work for the specific use-case my organization is trying to solve. It seems like it would work...I told my manager that I could always have a process that just sits around and downloads recordings when they are finished. I'll thinking I'll build a prototype tool that does that and see if that meets the requirements.
Anyway, thanks again for giving me a little more insight on how moonfire is intended to be used!

scottlamb, 2024-02-14:
Welcome!

> I told my manager that I could always have a process that just sits around and downloads recordings when they are finished.

A few related things that might also be handy for getting video live or near-live:

* existing API for getting chunks of video as they come in live: [`GET /api/cameras/<uuid>/<stream>/live.m4s
`](https://github.com/scottlamb/moonfire-nvr/blob/master/ref/api.md#get-apicamerasuuidstreamlivem4s)
* I've thought about (not sure if there's an issue for now) serving video in other forms: HLS, RTSP, and/or WebRTC.
* #40 open to just have a stream that just tells you when there's a new recording or other changes

I'm totally open to extending the API as needed, although due to my time constraints it's likely more of an "I'm happy to review/advise on designs and PRs" situation than an "I can implement this for you on your desired timeline" situation for anything large.

jonathan84clark, 2024-02-16:
Thanks for all the help! I was able to put together a script that can download .mp4 files using the web API. I have one other question. 
When I create a camera in the config file, a UUID seems to get auto-generated. I've seen UUIDs with hard drives before. If I recall, the UUID for a hard drive is created when the drive is formatted. Is the UUID for the camera just a randomly generated byte string? Is it just generated when the camera is added? Or does the UUID match up with certain characteristics of the stream?

scottlamb, 2024-02-16:
It's currently a v4 (random) uuid, assigned on camera creation [here](https://github.com/scottlamb/moonfire-nvr/blob/dbf6c2f4765a40aee8b2cb85243e7e14d8ed1de5/server/db/db.rs#L1910).

jonathan84clark, 2024-02-16:
Thanks for the information!

closed 2024-04-18 by scottlamb