scottlamb/moonfire-nvr #319: moonfire-nvr.toml file "dbDir" is not working

issueser, 2024-06-16:
**Describe the bug**
moonfire-nvr.toml file "dbDir" is not working
﻿
**To Reproduce**
1.Edit "/etc/moonfire-nvr.toml" according to the description in "config.md", first line is [dbDir = "/mnt/data/nvr/db"]
2.sudo -u moonfire-nvr moonfire-nvr config
2024-06-16T10:42:12.874057 ERROR main moonfire_nvr: exiting due to error err=NOT_FOUND: db dir /var/lib/moonfire-nvr/db not found; try running moonfire-nvr init
3.run command [sudo -u moonfire-nvr moonfire-nvr init] result:
2024-06-16T10:57:29.790579  INFO main moonfire_nvr::cmds: Opening /var/lib/moonfire-nvr/db/db in Create mode with SQLite version 3.44.0
2024-06-16T10:57:29.903563  INFO main moonfire_nvr::cmds::init: Database initialized.
﻿
**Expected behavior**
SQLite dir : /mnt/data/nvr/db
false "/var/lib/moonfire-nvr/db/db"  x
true "/mnt/data/nvr/db/db"  √
﻿
**Screenshots**
Not to that point
﻿
**Server (please complete the following information):**
sudo -u moonfire-nvr moonfire-nvr --version
Version: v0.7.16
filename: moonfire-nvr-v0.7.16-aarch64
OS:armbian
﻿
**Camera (please complete the following information):**
Not to that point
﻿
**Desktop (please complete the following information):**
Chromium 102.0.5005.167
﻿
**Smartphone (please complete the following information):**
Not to that point
﻿
**Additional context**
Sorry, I'm viewing the document through translation read. For "At the top level, before any [[bind]] lines, the following keys are understood", I understand it as the first line setting, or setting before pasting
================
[[binds]]
ipv4 = "0.0.0.0:8080"
allowUnauthenticatedPermissions = { viewVideo = true }
﻿
[[binds]]
unix = "/var/lib/moonfire-nvr/sock"
ownUidIsPrivileged = true
================
Is my understanding wrong?
Searching for "repo:scottlamb/moonfire-nvr dbDir" found that "dbDir" keyword is only mentioned in ref/config.md, or the documentation is missing updates? Please help me look into it, thank you.
﻿
( Manually copying db files after executing SQL init is also not available
﻿

issueser, 2024-06-16:
Sorry, I don't know why there was a problem with the layout

scottlamb, 2024-06-16:
> first line is `[dbDir = "/mnt/data/nvr/db"]`

I think you just have to remove the `[` and `]`, and it should work.

> Sorry, I don't know why there was a problem with the layout

For some reason all the blank lines had a U+FEFF (zero width non-breaking space, aka byte order mark) in them. I edited your comment to remove those and switched from `================` to <tt>```</tt> around the config snippet at the bottom, and it looks a bit better now.

issueser, 2024-06-16:
thank for you edited.
first line:`dbDir = "/mnt/data/nvr/db"` , `[` and `]` The original intention was to use it as a boundary.
full "/etc/moonfire-nvr.toml":
```
dbDir = "/mnt/data/alist/nvr/db"

[[binds]]
ipv4 = "0.0.0.0:8080"
allowUnauthenticatedPermissions = { viewVideo = true }

[[binds]]
unix = "/var/lib/moonfire-nvr/sock"
ownUidIsPrivileged = true
```
I tried again but it still didn't work.
init and config Still using `/var/lib/moonfire-nvr/db/db`.
_INFO main moonfire_nvr::cmds: Opening **/var/lib/moonfire-nvr/db/db** in ReadWrite mode with SQLite version 3.44.0_
```
# sudo -u moonfire-nvr moonfire-nvr config
2024-06-17T04:58:57.988461  INFO main moonfire_nvr::cmds: Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.44.0
2024-06-17T04:58:58.048116  INFO main moonfire_db::db: Loading video sample entries
2024-06-17T04:58:58.049107  INFO main moonfire_db::db: Loaded 0 video sample entries
2024-06-17T04:58:58.049209  INFO main moonfire_db::db: Loading sample file dirs
2024-06-17T04:58:58.049568  INFO main moonfire_db::db: Loaded 0 sample file dirs
2024-06-17T04:58:58.049676  INFO main moonfire_db::db: Loading cameras
2024-06-17T04:58:58.051311  INFO main moonfire_db::db: Loaded 0 cameras
2024-06-17T04:58:58.051410  INFO main moonfire_db::db: Loading streams
2024-06-17T04:58:58.051785  INFO main moonfire_db::db: Loaded 0 streams
2024-06-17T04:59:01.743079  INFO main flush{flush_count=0 reason="drop"}: moonfire_db::db: flush complete:  no recording changes
```
(After entering the configuration interface, I proactively terminated it.)