scottlamb/moonfire-nvr #322: I am getting single frames insted of a video stream.

Tuhinpaul5, 2024-07-04:
**Describe the bug**
I am getting 0 sec frames insted of a live feed or recording.

**Screenshots**
[image]

**Logs**
```
sudo moonfire-nvr run
2024-07-04T12:37:09.335972  INFO main moonfire_nvr::cmds: Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.44.0
2024-07-04T12:37:09.349844  INFO main moonfire_db::db: Loading video sample entries
2024-07-04T12:37:09.349904  INFO main moonfire_db::db: Loaded 2 video sample entries
2024-07-04T12:37:09.349911  INFO main moonfire_db::db: Loading sample file dirs
2024-07-04T12:37:09.349986  INFO main moonfire_db::db: Loaded 1 sample file dirs
2024-07-04T12:37:09.349991  INFO main moonfire_db::db: Loading cameras
2024-07-04T12:37:09.350038  INFO main moonfire_db::db: Loaded 1 cameras
2024-07-04T12:37:09.350043  INFO main moonfire_db::db: Loading streams
2024-07-04T12:37:09.350087  INFO main moonfire_db::db: Loaded 3 streams
2024-07-04T12:37:09.350093  INFO main moonfire_db::db: Loading recordings for camera cam1 stream Main
2024-07-04T12:37:09.350167  INFO main moonfire_db::db: Loaded 5 recordings for camera cam1 stream Main
2024-07-04T12:37:09.350173  INFO main moonfire_db::db: Loading recordings for camera cam1 stream Sub
2024-07-04T12:37:09.350188  INFO main moonfire_db::db: Loaded 5 recordings for camera cam1 stream Sub
2024-07-04T12:37:09.350192  INFO main moonfire_db::db: Loading recordings for camera cam1 stream Ext
2024-07-04T12:37:09.350205  INFO main moonfire_db::db: Loaded 5 recordings for camera cam1 stream Ext
2024-07-04T12:37:09.350212  INFO main moonfire_nvr::cmds::run: Database is loaded.
2024-07-04T12:37:09.353803  INFO main moonfire_nvr::cmds::run: Directories are opened.
2024-07-04T12:37:09.353828  INFO main moonfire_nvr::cmds::run: Resolved timezone: Asia/Kolkata
2024-07-04T12:37:09.353843  INFO main moonfire_nvr::cmds::run: Starting syncer for path /home/ratnadeep/moon_cam1/
2024-07-04T12:37:09.353891  INFO main syncer{path=/home/ratnadeep/moon_cam1/}: moonfire_db::writer: initial rotation
2024-07-04T12:37:09.357359  INFO main syncer{path=/home/ratnadeep/moon_cam1/}:flush{flush_count=0 reason="synchronous deletion"}: moonfire_db::db: flush complete: 
/home/ratnadeep/moon_cam1/: added 0B in 0 recordings (), deleted 1M 537K 947B in 15 (3/3046, 3/3047, 3/3048, 3/3049, 3/3050, 2/3043, 2/3044, 2/3045, 2/3046, 2/3047, 1/3273, 1/3274, 1/3275, 1/3276, 1/3277), GCed 0 recordings ().
2024-07-04T12:37:09.364263  INFO main syncer{path=/home/ratnadeep/moon_cam1/}:flush{flush_count=1 reason="synchronous garbage collection"}: moonfire_db::db: flush complete: 
/home/ratnadeep/moon_cam1/: added 0B in 0 recordings (), deleted 0B in 0 (), GCed 15 recordings (2/3046, 2/3045, 3/3050, 2/3047, 1/3277, 2/3043, 1/3275, 1/3276, 3/3049, 2/3044, 1/3274, 1/3273, 3/3047, 3/3048, 3/3046).
2024-07-04T12:37:09.364389  INFO sync-1 syncer{path=/home/ratnadeep/moon_cam1/}: moonfire_db::writer: starting
2024-07-04T12:37:09.364446  INFO s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::cmds::run: starting
2024-07-04T12:37:09.364488  INFO        main moonfire_nvr::cmds::run: no LISTEN_FDs
2024-07-04T12:37:09.364478  INFO  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::cmds::run: starting
2024-07-04T12:37:09.364489  INFO s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:09.364502  INFO  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:09.364505  INFO  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::cmds::run: starting
2024-07-04T12:37:09.364540  INFO  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:09.376649  INFO        main moonfire_nvr::cmds::run: Ready to serve HTTP requests
2024-07-04T12:37:09.523948  WARN  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
2024-07-04T12:37:09.537967  WARN  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
2024-07-04T12:37:09.552767  WARN s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
2024-07-04T12:37:10.524149  INFO  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:10.538085  INFO  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:10.552978  INFO s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.0.166:554/stream1
2024-07-04T12:37:10.689907  WARN  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
2024-07-04T12:37:10.701523  WARN  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
2024-07-04T12:37:10.707313  WARN s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=INVALID_ARGUMENT: pts not monotonically increasing; got 0 then 0
^C2024-07-04T12:37:10.941346  INFO        main moonfire_nvr::cmds::run: Received SIGINT; shutting down gracefully. Send another SIGINT or SIGTERM to shut down immediately.
2024-07-04T12:37:10.941437  INFO        main moonfire_nvr::cmds::run: Shutting down streamers and syncers.
2024-07-04T12:37:11.690104  INFO  s-cam1-sub streamer{stream="cam1-sub"}: moonfire_nvr::streamer: shutting down
2024-07-04T12:37:11.701670  INFO  s-cam1-ext streamer{stream="cam1-ext"}: moonfire_nvr::streamer: shutting down
2024-07-04T12:37:11.707445  INFO s-cam1-main streamer{stream="cam1-main"}: moonfire_nvr::streamer: shutting down
2024-07-04T12:37:11.707900  INFO        main moonfire_nvr::cmds::run: Waiting for HTTP requests to finish.
2024-07-04T12:37:11.707914  INFO        main moonfire_nvr::cmds::run: Waiting for TEARDOWN requests to complete.
2024-07-04T12:37:11.707921  INFO        main moonfire_nvr::cmds::run: Exiting.
2024-07-04T12:37:11.718839  INFO        main flush{flush_count=2 reason="drop"}: moonfire_db::db: flush complete: 
/home/ratnadeep/moon_cam1/: added 492K 303B in 6 recordings (3/3051, 3/3052, 1/3278, 1/3279, 2/3048, 2/3049), deleted 0B in 0 (), GCed 0 recordings ().
```

**Camera (please complete the following information):**
  - Camera manufacturer and model: TAPO c220

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser : Chrome

scottlamb, 2024-07-06:
If you're able to, it'd really help to get a packet capture (e.g. with [Wireshark](https://www.wireshark.org/)) of the communication with your camera: with Moonfire, and/or with some other software that is successfully getting video.

tklkalok, 2025-04-05:
@Tuhinpaul5 @scottlamb I am having the same issue as well, using Tapo C120 camera.

@scottlamb If you could, can you please tell us how to grab the data you needed? I saw you mentioned wireshark but I am not sure if I know how to capture the data you wanted..

Thank you!

[image]