scottlamb/moonfire-nvr #327: Only getting 2 minutes of recording and a single row entry in the GUI

nh2, 2024-10-30:
First, thanks for this software, I'm excited to try it out with my Axis M3024-L camera!

This is probably some configuration problem on my side, but I can't figure it out:

I'm getting only a single "row" in the table of recordings:

[image]

The recorded streams seem to always grow up to 2 minutes, and at the end of each minute they advance by 1 minute.

I'm expecting to see multiple entries as in your README screenshot of the recordings list: https://github.com/scottlamb/moonfire-nvr/blob/3739e634abe10f2e5b16e5d6e5e4b6fa6863abf5/screenshots/list.png

And the stream recordings on disk don't seem to grow beyond the following:

```
4.4M	sample-dir/000000010000005c
2.1M	sample-dir/000000010000005d
393K	sample-dir/000000020000005c
512	sample-dir/000000020000005d
4.5K	sample-dir/meta
```

I've set no recording limits in the config:

[image]

Am I missing something?

(Using `moonfire-nvr-v0.7.17-x86_64` on Linux.)

nh2, 2024-10-30:
The logs say

```
sample-dir: added 4M 481K 842B in 1 recordings (1/94), deleted 4M 350K 643B in 1 (1/93), GCed 1 recordings (2/92)
```

but I don't get why it does that deleting/GCing, given that I've set no limit in the config.

nh2, 2024-10-30:
I have the suspicion that the default `0` doesn't mean "unlimited" but somehow `2 * flush_if_sec` or something like that.

If I set a non-`0` `limit`, e.g. to `100G`, it works as expected.

scottlamb, 2026-02-16:
You're right that 0 means 0 bytes of flushed, undeleted recordings (plus some bytes of unflushed recordings and some bytes of recordings that are in the process of being deleted), rather than unlimited.

What would you expect "unlimited" to do when running out of disk space?

Currently recording will essentially hang, which is not so great. That's why it enforces that the limits for all the streams in a sample file directory sum to no more than the available disk space in that directory and why the install guide recommends going a bit lower:

https://github.com/scottlamb/moonfire-nvr/blob/424d61f2b5c23d182a1f1dc6ace9e5c7632fa7c3/guide/install.md#L283-L298