scottlamb/moonfire-nvr #328: Feature: Simple text-based config

nh2, 2024-10-30:
Currently `moonfire-nvr` has a mix of a config file and then more config in the DB.

It would be great if all config could optionally be done declaratively in a text file.

This make it easier to:

* check the config into version control (e.g. git), and diff it
* use it in declarative systems such as NixOS Linux
  * see e.g. [this](https://discourse.nixos.org/t/frigate-nvr-nix-config/35882/2) for a declarative [`frigate`](https://github.com/blakeblackshear/frigate) config, it would be great to be able to do this with Moonfire as well
* have users post their full config in guides and issues, instead of having to provide screenshots
* increase simplicity and scriptability: Directly invoke `moonfire-nvr run --config myconfig.toml` with a single invocation, no separate `init` or `config` calls needed; they'd be done automatically; no need to navigate `curses` UIs (I appreciate that it's a cool feature that's nice for some groups of users, but many others might find the "just invoke 1 binary on 1 config file" approach even easier)

---

I understand that you also want to provide the feature to change settings, cameras, etc. via an API. For that, a DB is of course more suitable than a read-only configuration text file.

For that I propose to control the behaviour with some options:

* One to say "use this config file to initialise the DB; afterwards, config values in the DB persist and get used after restart".
* One to say "just use the config files; further configs done at run-time apply in-memory but get wiped on next restart".
* One to say "just use the config files, read-only; reject any config change via the API or similar".

Alibirb, 2025-01-10:
Moonfire having a text-based config file would also be very convenient for the NVR system I'm working on, which wraps Moonfire and uses it as the recording backend. It would be a lot easier to automatically configure Moonfire if it just took a config file.

scottlamb, 2025-02-22:
I'm skeptical of going this route. In particular, regarding these modes:
 
> * One to say "just use the config files; further configs done at run-time apply in-memory but get wiped on next restart".
> * One to say "just use the config files, read-only; reject any config change via the API or similar".

How would these modes work if the database is inconsistent with the config file? Moonfire has a very different design than Frigate with stricter ideas of database integrity. For example, each camera has an internal id (used in the SQLite database and sample file directories) and a uuid. If a user changes one or both between runs, what is expected to happen?

> no need to navigate curses UIs (I appreciate that it's a cool feature that's nice for some groups of users, but many others might find the "just invoke 1 binary on 1 config file" approach even easier)

I do want to get rid of this in favor of providing only a config HTTP API (and a web UI on top of it).

> Moonfire having a text-based config file would also be very convenient for the NVR system I'm working on, which wraps Moonfire and uses it as the recording backend. It would be a lot easier to automatically configure Moonfire if it just took a config file.

I'd love to hear more about your system sometime. I would certainly like such things to be possible atop the API.

Alibirb, 2025-04-11:
> I'd love to hear more about your system sometime. I would certainly like such things to be possible atop the API.

The basic idea of [ClusterVMS](https://github.com/ClusterVMS) is to make a modular VMS framework/ecosystem where different components can be swapped out to fit different use cases. There's a lot that goes into a VMS, and a wide range of use cases. Some, like Moonfire or Frigate, seem focused mainly on a particular aspect, and may fall short in other areas (at least from the perspective of a user with a particular use case). So the idea is to break it up into pieces, and let each piece work somewhat independently. The UI is separate from the backend, the recording is separate from authentication and user management, et cetera. If it fundamentally falls flat in one area for someone's use case, they could just write their own replacement for the relevant component, rather than need to make a whole new system.

I still need to get around to creating a Discord and some proper documentation, but [the main repo](https://github.com/ClusterVMS/clustervms/tree/main) has a README that explains the project, its goals, philosophy, et cetera.

As for configuring Moonfire, I think either a text-based config or an HTTP API would be suitable for our purposes. I generally like text-based configs, but an HTTP API would likely have the benefit of allowing runtime configuration changes without needing a restart (at least for some settings).