scottlamb/moonfire-nvr #333: h265 not working: "unable to get first frame"

Suxsem, 2025-01-29:
Hi,

i'm trying the new h265 support feature but I'm getting this error:

[image]

VLC is able to play the stream without any issue.

[image]

Thank you

scottlamb, 2025-01-29:
The problem is pretty clear from the screenshot, but it'd be nice to get the actual SPS bytes for test data to confirm the fix. Would it be possible to send me a packet capture?

Suxsem, 2025-01-29:
Sure! Can you guide me in doing so? Is it possible to export it from VLC?

scottlamb, 2025-01-29:
There might be some way to get VLC to export the needed data, but I was thinking via [Wireshark](https://www.wireshark.org/):

1. Start wireshark on the network interface you're using, with a filter like `host 192.168.11.159`
2. Connect via either Moonfire or VLC; it's fine if it errors out right away like this.
3. Stop/save the capture, and email it to me.

In the next release, I'll also improve Moonfire's logging a bit to include the actual SPS bytes itself.

Suxsem, 2025-01-30:
@scottlamb sent! thank you!

Suxsem, 2025-01-30:
@scottlamb 

just for curiosity I tried to extract the SPS from the wireshark dump and I found it in the DESCRIBE response, here are the base64 strings of the VPS, SPS and the PPS:

QAEMAf//AWAAAAMAAAMAAAMAAAMAuqwJ
QgEBAWAAAAMAAAMAAAMAAAMAuqABICAFEf5a7kSIi/Lc1AQEBAI=
RAHA8oSJAzJA

hope it helps!

scottlamb, 2025-01-31:
Thanks! This `inter_ref_pic_set_prediction_flag` logic is quite intricate, but I'm polishing up something that appears to work for your camera's SPS.

For my notes in test cases, changelogs, etc.: what is the make/model/version of the camera you're using?

Suxsem, 2025-01-31:
Sure! It's the Xiaomi YI Pro 2K Home running the great https://github.com/roleoroleo/yi-hack-Allwinner-v2 firmware. This firmware is quite popular and supports most of Xiaomi cameras so by supporting my camera you are actually adding support for a lot of models.

scottlamb, 2025-02-01:
This should be fixed with Moonfire NVR v0.7.20.

closed 2025-02-01 by scottlamb

Suxsem, 2025-02-01:
it works!! Thank you!

nalonsor1, 2025-02-02:
Hello: I am nalonsor again testing tenda cp3pro h-265 with 0.7.20
this is what i got

2025-02-02T02:56:08.444690  INFO main moonfire_db::db: Loaded 1 cameras
2025-02-02T02:56:08.444840  INFO main moonfire_db::db: Loading streams
2025-02-02T02:56:08.445057  INFO main moonfire_db::db: Loaded 1 streams
2025-02-02T02:56:08.445203  INFO main moonfire_db::db: Loading recordings for camera CP3PTerraza stream Main
2025-02-02T02:56:08.445385  INFO main moonfire_db::db: Loaded 0 recordings for camera CP3PTerraza stream Main
2025-02-02T02:56:08.445555  INFO main moonfire_nvr::cmds::run: Database is loaded.
2025-02-02T02:56:08.446518  INFO main moonfire_nvr::cmds::run: Directories are opened.
2025-02-02T02:56:08.446654  INFO main moonfire_nvr::cmds::run: Resolved timezone: Europe/Madrid
2025-02-02T02:56:08.446775  INFO main moonfire_nvr::cmds::run: Starting syncer for path /media/nvr/sample
2025-02-02T02:56:08.446967  INFO main syncer{path=/media/nvr/sample}: moonfire_db::writer: initial rotation
2025-02-02T02:56:08.447649  INFO main syncer{path=/media/nvr/sample}:flush{flush_count=0 reason="synchronous deletion"}: moonfire_db::db: flush complete:  no recording changes
2025-02-02T02:56:08.447925  INFO main moonfire_nvr::cmds::run: no LISTEN_FDs
2025-02-02T02:56:08.452566  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::cmds::run: starting
2025-02-02T02:56:08.452743  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.40.125/ch=1?subtype=1
2025-02-02T02:56:08.454369  INFO             sync-1 syncer{path=/media/nvr/sample}: moonfire_db::writer: starting
2025-02-02T02:56:08.458840  INFO               main moonfire_nvr::cmds::run: Ready to serve HTTP requests
2025-02-02T02:56:09.160594  WARN s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN: unable to get first frame
caused by: FU has inconsistent NAL type: Header { unit_type: IdrWRadl, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 } then Header { unit_type: TrailR, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 }

conn: 192.168.40.124:40032(me)->192.168.40.125:554@2025-02-02T01:56:08.455271827Z
stream: TCP, interleaved channel ids 0-1
ssrc: 1689aea5
seq: 27644
pkt: 2934@2025-02-02T01:56:09.160384398Z
2025-02-02T02:56:10.160947  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.40.125/ch=1?subtype=1
2025-02-02T02:56:10.937029  WARN s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN: unable to get first frame
caused by: FU has inconsistent NAL type: Header { unit_type: IdrWRadl, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 } then Header { unit_type: TrailR, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 }

conn: 192.168.40.124:40042(me)->192.168.40.125:554@2025-02-02T01:56:10.164830969Z
stream: TCP, interleaved channel ids 0-1
ssrc: fdd1c4a0
seq: 55729
pkt: 2934@2025-02-02T01:56:10.936972102Z
2025-02-02T02:56:11.937411  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.40.125/ch=1?subtype=1
2025-02-02T02:56:11.937580  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: waiting up to 58.999401897s for TEARDOWN or expiration of 1 stale sessions
2025-02-02T02:56:13.804833  INFO s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: done waiting; no more stale sessions
2025-02-02T02:56:14.273035  WARN s-CP3PTerraza-main streamer{stream="CP3PTerraza-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN: unable to get first frame
caused by: FU has inconsistent NAL type: Header { unit_type: IdrWRadl, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 } then Header { unit_type: TrailR, nuh_layer_id: 0, nuh_temporal_id_plus1: 1 }

scottlamb, 2025-02-02:
@nalonsor1 Could you email me a packet capture? I have an idea what may be happening but would like to confirm it before writing a bunch of code.