scottlamb/moonfire-nvr #337: Build fails on FreeBSD 14.2

jaredwolff, 2025-05-08:
**Describe the bug**

Build fails with this error:

```shell
error[E0308]: mismatched types
   --> base/error.rs:123:13
    |
104 |         match self {
    |               ---- this expression has type `&Errno`
...
123 |             Error::EOPNOTSUPP => ErrorKind::Unimplemented,
    |             ^^^^^^^^^^^^^^^^^ expected `&Errno`, found `Errno`
    |
help: consider dereferencing to access the inner value using the Deref trait
    |
104 |         match &*self {
    |               ++
```

**To Reproduce**

1. Clone repository on a FreeBSD machine
2. Change to `server` directory
3. Run `cargo build --release`

**Expected behavior**

To build successfully

**Additional context**

Not sure why this is happening. Happy to pop a PR in to conditionally change the line to this if the target is FreeBSD: 

```diff
diff --git a/server/base/error.rs b/server/base/error.rs
index 9116e52..2d38f4a 100644
--- a/server/base/error.rs
+++ b/server/base/error.rs
@@ -120,6 +120,9 @@ impl ToErrKind for nix::Error {
             Error::EBADF | Error::EFAULT => ErrorKind::InvalidArgument,
             Error::EINTR | Error::EAGAIN => ErrorKind::Aborted,
             Error::ENOENT | Error::ENODEV => ErrorKind::NotFound,
+            #[cfg(target_os = "freebsd")]
+            &Error::EOPNOTSUPP => ErrorKind::Unimplemented,
+            #[cfg(not(target_os = "freebsd"))]
             Error::EOPNOTSUPP => ErrorKind::Unimplemented,
             _ => ErrorKind::Unknown,
         }
```

scottlamb, 2025-05-08:
That is confusing. I'll try to set up a FreeBSD when I get a chance. What version of Rust are you building with?

jaredwolff, 2025-05-08:
Yea, I'm as perplexed as you are! Why is only THAT value throwing the error.

```
root@moonfire:~ # rustc --version
rustc 1.86.0 (05f9846f8 2025-03-31)
```

I tried upgrading and searching the `nix` crate to see if there were any related issues but couldn't find any.

scottlamb, 2025-05-09:
Apparently a handful of `nix::Errno::BLAH` things are associated constants rather than variants, and which differ by platform:

* [FreeBSD](https://github.com/nix-rust/nix/blob/996db47d542ae20f09eb344b9fcb88c40ae38e3d/src/errno.rs#L1782-L1787)
* [Linux](https://github.com/nix-rust/nix/blob/996db47d542ae20f09eb344b9fcb88c40ae38e3d/src/errno.rs#L1293-L1297)

I guess the right thing is probably to do `match *self` and then it will probably work for any of these regardless of platform.

PR #339 (2025-06-19): Addressing FreeBSD compile issue
Addressing the compile issue on FreeBSD. Fixes #337
closed 2025-06-19 by scottlamb