scottlamb/moonfire-nvr #343: SourceBuffer error in live view

hn, 2025-08-19:
**Describe the bug**
After some time (roughly between 2 and 60 seconds), the live image freezes and the message “SourceBuffer error” is displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'live view'
2. Click on '1' or '2x2' view or do nothing (doesn't matter)
3. Wait some time
4. See error overlay

**Expected behavior**
Live streams without any error or auto-resuming after an error.

**Server (please complete the following information):**
  - moonfire-nvr-v0.7.21-x86_64 (same experience with older versions)
  - No error in standard moonfire nvr log
  - Javascript browser console:
```
index-BbqwXmj9.js:75 CameraSouth: starting stream: sourceopen
index-BbqwXmj9.js:75 CameraWest: starting stream: sourceopen
index-BbqwXmj9.js:75 CameraSouth: trimming 0-5.096949
index-BbqwXmj9.js:75 CameraWest: trimming 0-5.104308
index-BbqwXmj9.js:75 CameraSouth: trimming 3.333317-8.43279
index-BbqwXmj9.js:75 CameraWest: trimming 4.171-9.292428
index-BbqwXmj9.js:75 CameraSouth: trimming 6.666616-11.721595
index-BbqwXmj9.js:75  CameraSouth: aborting due to SourceBuffer error
(anonymous) @ index-BbqwXmj9.js:75
(anonymous) @ index-BbqwXmj9.js:75
index-BbqwXmj9.js:75 CameraSouth: stopping stream: SourceBuffer error
index-BbqwXmj9.js:75  CameraSouth: bufUpdateEnd in state error
(anonymous) @ index-BbqwXmj9.js:75
index-BbqwXmj9.js:75 CameraWest: trimming 8.342-13.442284
index-BbqwXmj9.js:75 CameraWest: trimming 12.48-17.564951
index-BbqwXmj9.js:75 CameraWest: trimming 16.651-21.700509
index-BbqwXmj9.js:75 CameraWest: trimming 20.788-25.854282
index-BbqwXmj9.js:75 CameraWest: trimming 24.926-29.988833
index-BbqwXmj9.js:75 CameraWest: trimming 29.064-34.152779
index-BbqwXmj9.js:75 CameraWest: trimming 33.200998-38.312312
index-BbqwXmj9.js:75 CameraWest: trimming 37.338998-42.374517
index-BbqwXmj9.js:75 CameraWest: trimming 41.475998-46.524138
index-BbqwXmj9.js:75 CameraWest: trimming 45.613998-50.660581
index-BbqwXmj9.js:75 CameraWest: trimming 49.750998-54.834791
index-BbqwXmj9.js:75 CameraWest: trimming 53.888998-58.996078
index-BbqwXmj9.js:75 CameraWest: trimming 58.025998-63.090463
index-BbqwXmj9.js:75 CameraWest: trimming 62.163998-67.201926
index-BbqwXmj9.js:75 CameraWest: trimming 66.334993-71.373879
index-BbqwXmj9.js:75  CameraWest: aborting due to SourceBuffer error
(anonymous) @ index-BbqwXmj9.js:75
(anonymous) @ index-BbqwXmj9.js:75
index-BbqwXmj9.js:75 CameraWest: stopping stream: SourceBuffer error
index-BbqwXmj9.js:75  CameraWest: bufUpdateEnd in state error
(anonymous) @ index-BbqwXmj9.js:75
(anonymous) @ index-BbqwXmj9.js:75
(anonymous) @ index-BbqwXmj9.js:75
(anonymous) @ index-BbqwXmj9.js:75
```

**Camera (please complete the following information):**
  - All camera models, doesn't seem to matter

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Edge, Firefox
 - Version : Current, e.g. Edge Version 138.0.3351.109 (older Edge/Firefoxes as well)

**Additional context**
The 'trimming' messages in the Javascript console look odd, what's the meaning of this?

scottlamb, 2026-02-16:
fwiw, this works far better on Chrome. I don't know why. But I tried it on Firefox just now and was able to see an error not long after starting the stream. Not exactly the same error you quoted above; I got `Media resource blob:https://nvr.home.slamb.org/92a44c90-0805-496e-beaf-e253212a8e77 could not be decoded.` then `courtyard: aborting due to InvalidStateError while appending buffer`. I don't know if that's due to browser version (Firefox 147.0.4 in my case) or a different underlying cause.

[image]

When I view the same segment of video through the list interface, it plays just fine. So maybe it's something about the way we're encoding into segmented `.mp4` files (aka `.m4s`), or maybe it's how we're managing the MSE sequence of operations, or maybe Firefox is just pickier when using the HTML Media Source API than via old-fashioned `.mp4` file. (iirc the codepath goes through completely different demuxing libraries.)

> The 'trimming' messages in the Javascript console look odd, what's the meaning of this?

The JavaScript code is expected to manage the size of the source buffer to avoid exceeding a quota limit (and annoyingly I don't think there's a way to see what the limit is from the Javascript except by trial and error). This code tries to do that. In a lines like these,

```
index-BbqwXmj9.js:75 CameraWest: trimming 62.163998-67.201926
index-BbqwXmj9.js:75 CameraWest: trimming 66.334993-71.373879
```

the first number is the current starting time in seconds since the stream started, and the second is the intended new starting time. The overlap from line to line I think are because the calls currently don't line up precisely with frames and the browser rounds down. Although the overlap here (67.201926 - 66.334993 ~= 0.87 seconds) seems too high if your stream is > 2 fps; I don't know if that's significant.

https://github.com/scottlamb/moonfire-nvr/blob/424d61f2b5c23d182a1f1dc6ace9e5c7632fa7c3/ui/src/Live/LiveCamera.tsx#L280-L285

https://github.com/scottlamb/moonfire-nvr/blob/424d61f2b5c23d182a1f1dc6ace9e5c7632fa7c3/ui/src/Live/LiveCamera.tsx#L301-L308

I don't know what's going on. Maybe the thing to do is to cobble together some Javascript that makes this happen with the `view.m4s` endpoint rather than the `live.m4s` endpoint for exact reproducibility, then tweak it to answer questions: Does it happen when the playback hits a certain time/byte position? When appending a certain segment of buffer? Is it because we've trimmed too much?

hn, 2026-02-18:
Thanks! I'll test if I have time to build moonfire-nvr or you release a new version, whatever comes first.

hn, 2026-02-19:
Couldn't test .28 since you were too fast with .29 🙂 The latter looks promising. Since it's night here right now, all streams have switched to a greyish night view with lower bandwidth, so it's probably not a typical scenario.

Am I confused, or has the menu bar at the top switched to blue?